hzeller/rpi-rgb-led-matrix #1708: glitch on P5 panel (32x64)

Stalker-moment, 2024-09-18:
hello please help me, I have a glitch on my P5 panel (32x64), 2 chain length. How to solve this glitch?
I am using classic configuration on raspberry pi 3B for this case because my panel does not have E pin. Is there a hardware/code solution to eliminate this glitch effect? Thanks

https://github.com/user-attachments/assets/f604a68b-2aa3-4502-8d27-aff27531db39

bitslip6, 2024-10-27:
Looks like a signal issue. Try gpio slowdown command line option. could be your cables are too long.

marcmerlin, 2024-11-15:
You can also use different ribbon cables, but ultimately there is a ribbon length issue. gpio slowdown will help, to a point.
Please re-open if appropriate, but otherwise not much more help we can give, it's really looking like a hardware issue.

closed 2024-11-15 by marcmerlin