hzeller/rpi-rgb-led-matrix #1711: New 64 x 64 panel - alternate groups of 16 rows displaying

bidmangler, 2024-09-29:
[image]
I'm hoping for some pointers about where to start troubleshooting.

I'm working with these panels from a company called "Seengreat" - https://seengreat.com/wiki/74/rgb-matrix-p3-0-64x64 . I've also built a bunch of the level-shifting adapter boards from this project so I can use multiple chains.  From the panel documentation, I've deduced that its ABCDE addressing - with E on Pin 4 of the Hub75 connector. I set the jumpers on the adapter board accordingly.

Just trying to run a basic demo at this stage with just one panel connected to chain 1 but get the problem you can see in the photo. I've tried different multiplexing options but no joy.

Any ideas? How would I be best to go about troubleshooting this? I did check all the info on the page - and looked for relevant issues but nothing jumps out at me I'm afraid. 

Thanks folks!

tidyhf, 2025-01-25:
I'm having a similar issue, I have bridged E to 8 and also tried braiding E to 16 but both of them have this same result.  If I use demo 10 I can see a tiny bit of light on the unlit sections.

davemaster, 2025-01-26:
> [image] I'm hoping for some pointers about where to start troubleshooting.
> 
> **Update: Solved. the E address line was on Pin 8 after all - I was just counting the pin numbers on the Hub75 connector wrong.**
> 
> I'm working with these panels from a company called "Seengreat" - https://seengreat.com/wiki/74/rgb-matrix-p3-0-64x64 . I've also built a bunch of the level-shifting adapter boards from this project so I can use multiple chains. From the panel documentation, I've deduced that its ABCDE addressing - with E on Pin 4 of the Hub75 connector. I set the jumpers on the adapter board accordingly.
> 
> Just trying to run a basic demo at this stage with just one panel connected to chain 1 but get the problem you can see in the photo. I've tried different multiplexing options but no joy.
> 
> Any ideas? How would I be best to go about troubleshooting this? I did check all the info on the page - and looked for relevant issues but nothing jumps out at me I'm afraid.
> 
> Thanks folks!

Greetings,

You share an image, but not the command used to produce that image. Please, share the command, and a picture or BETTER, a video of output.