hzeller/rpi-rgb-led-matrix #1712: display for 4x3 matrix panel

dhiksha10, 2024-09-30:
I am experiencing issues while using a 4x3 matrix display connected to a single Adafruit HAT. The display shows two different colors inconsistently. Additionally, when I attempt to increase the panel number to 12,(i.e --led-chain=12) there is no display output at all.

Adafruit HAT Model: regular
Matrix Display Model: P10 (32x16)
Wiring Details: Daisy chaining
Command : sudo python3 runtext.py  --led-cols=32  --led-rows=16  --led-chain=8  --led-no-hardware-pulse=LED_NO_HARDWAREPULSE  --text="Dhiksha" --led-multiplexing=3  --led-slowdown-gpio=2  --led-row-addr-type=2  --led-parallel=3  --led-pixel-mapper=U-mapper
Power Supply Specifications: 5V 40 A
Screenshots:
[image]

Environment
Operating System: Raspberry Pi 3 
Library Version: RGB Matrix Library 
Python Version: Python 3.7

Please let me know if there are any troubleshooting steps or modifications I can try to resolve these issues.

davemaster, 2024-09-30:
Greetings, 

40 A is too short for almost x10 panels, each 7-10A, get an 80A power supply, or another 40A

board707, 2024-10-14:
@davemaster 
Each these panels eats about 4A at full white brightness, so the 40A power supply seems to be way to enough for 4x3 screen with a few letters on it

bitslip6, 2024-10-27:
Isn't the ada fruit hat only 1 port?  Why you have parallel set to 3?

marcmerlin, 2024-11-15:
I think you got multiple suggestions, there is no issue with the code or anything to fix that I can see.
Basically make sure you have enough power and for that many panels, use the electrodragon active 3 board with 3 channels, not a single channel adafruit board.

closed 2024-11-15 by marcmerlin