hzeller/rpi-rgb-led-matrix #1713: 128x128 led panel

jeroenverspeek, 2024-10-06:
Currently I am working on a 128x128 panel (bought at https://dutch.alibaba.com/product-detail/P1-875-RGB-pixel-panel-HD-1601164735294.html ) but I don't get it to work.

I use a Raspberrypi 5 and a Electrodragon driver board. The panels seem to have only ABC address lines, also the HUB75 contains a pin labeled N.

Has anyone experience with 128x128 panels ?

[image]

board707, 2024-10-14:
> The panels seem to have only ABC address lines, also the HUB75 contains a pin labeled N.

Please show the types of the chips on the panel - your photo doesn't clear enough to read it.

jeroenverspeek, 2024-10-14:
The chipset is CHIPONE ICN2038S and CHIPONE ICND2018AP.

Compared to e.g. the image on issue #956 the A and B pins are interchanged, so it looks like BAC-addressing (?) instead of ABC-addressing.

With a Raspberrypi 4 and the Electrodragon hat version 2 the lower half rows 0..63 can give light. The leds in the upper half of the panel are not giving any light.

I tried code modifications like described in issue #466:
if (col == columns_-4) io->SetBits(h.strobe); // ICN2038S : Bird Techs
tep

and many command line options but nothing seems to work so far.

board707, 2024-10-14:
For ICN2018 you need a ABC-type row addressing: `--led-row-addr-type=3`. If you suspect the "BAC" addressing - just try to swap A and B wires on HUB75 connector - you won't damage anything by doing this.
The driver ICN2038S usually doesn't need any specific settings to run.

jeroenverspeek, 2024-10-18:
With command 
sudo ~/rpi-rgb-led-matrix/examples-api-use/demo -D 0 --led-rows 64 --led-cols 128 --led-parallel 2 --led-chain 1 --led-multiplexing 0 --led-slowdown-gpio 3 --led-row-addr-type 3

I get a rotating square on rows 0-63 with some glitches. Rows 64-127 (register 2) stay black.

https://github.com/user-attachments/assets/b8fa5827-5e6f-4cd1-bf73-025402b16d52

jeroenverspeek, 2024-10-27:
When connecting 2 data cables to one panel I get a whole squalre, using the HUB75E1 and HUB75E2 ports.

With command
sudo ./demo -D 0 --led-rows 64 --led-cols 128 --led-parallel 2 --led-chain 1 --led-multiplexing 0 --led-slowdown-gpio 4 --led-row-addr-type 3 --led-pwm-bits 11 --led-brightness 40 --led-show-refresh

I guess the panel is actually two 128x64 subpanels mounted on one plate, but the subpanels are not internally chained. My panel has 2 power connections and 2 HUB75 ports. Since both HUB75 are needed as input it is not possible to chain the panel (and I need 6 for my led-cube!).

I found panels with 2 power connections and 4 HUB75 on internet (https://www.ledcontrolcard.com/product/p1-875-indoor-smd-128x128/ , https://www.linsnled.com/p1-875mm-indoor-led-display-module.html) which are more expensive but problably are chainable.

[image]

davemaster, 2024-10-27:
> When connecting 2 data cables to one panel I get a whole square, using the HUB75E1 and HUB75E2 ports.
> 
> With command sudo ./demo -D 0 --led-rows 64 --led-cols 128 --led-parallel 2 --led-chain 1 --led-multiplexing 0 --led-slowdown-gpio 4 --led-row-addr-type 3 --led-pwm-bits 11 --led-brightness 40 --led-show-refresh
> 
> I guess the panel is actually two 128x64 subpanels mounted on one plate, but the subpanels are not internally chained. My panel has 2 power connections and 2 HUB75 ports. Since both HUB75 are needed as input it is not possible to chain the panel (and I need 6 for my led-cube!).
> 
> I found panels with 2 power connections and 4 HUB75 on internet (https://www.ledcontrolcard.com/product/p1-875-indoor-smd-128x128/ , https://www.linsnled.com/p1-875mm-indoor-led-display-module.html) which are more expensive but problably are chainable.
> 
> ![20241027_080514](https://private-user-images.githubusercontent.com/86149258/380475654-3e7b8809-683e-413a-99a3-4334e7fb4c54.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAwMjY5MjMsIm5iZiI6MTczMDAyNjYyMywicGF0aCI6Ii84NjE0OTI1OC8zODA0NzU2NTQtM2U3Yjg4MDktNjgzZS00MTNhLTk5YTMtNDMzNGU3ZmI0YzU0LmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI3VDEwNTcwM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVhZjdmOTdiMTU3NTYyODJiMDk4ODgyNDk4ZjAyOGRkN2IwNzk1MzAzZTIzZmJlM2E4YTQyODBiMjU3NzAxNTImWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.Wp2qnmbNCjtngJvsWViJwfmgJ8Nzntis2ch_GoYpscw)

Greetings,

Your panel has two input ports? Share picture of both pls. Thanks in advance

jeroenverspeek, 2024-10-27:
Here is a picture of the complete panel. The hub port on the right is labeled HUB75E1, the port on the left is labeled HUB75E2. I connected HUB75E1 to Electrodragon hub75 port 1 and HUB75E2 to Electrodragon hub75 port 2.

[image]