hzeller/rpi-rgb-led-matrix #1714: Glitches after 4 Panels

AlX-Core, 2024-10-08:
Hello,
i have 6 Panels on a Pi4, but after 4 panels i got some artifacts.
 
U1 Chip:
[image]
Chain:
[image]
Artifacts:
https://github.com/user-attachments/assets/d2743d8c-affa-4de6-9365-b06e1545f815

What i tryed: gpio slowdown 1-5. The Video is with 5. If i make it lower it gets worse.
Different panel-types, row-addr-type, multiplexing.

I checkt every pin connection. And swaped the panels. Same Issue.
For power i used a bench power supply at 5v. I checked with my multimeter an it says 4.9V. I use brightness 20-40% and it looks like that the higher the brightness the worse it gets. But the power supply dosnt reach its maximum current.

I also testet the panels with an esp32 and they worked fine. No glitches.

davemaster, 2024-10-08:
Greetings,

Share also, your command, what options are You using, Multiplexing ????

AlX-Core, 2024-10-08:
In the Video i just use slowdown GPIO 5, rows64, Cols64, chain6. Ist the best Setting i get so far.

AlX-Core, 2024-10-08:
All other options are default.

On the Pi4(8gb) i have Pi os lite 64Bit.

davemaster, 2024-10-08:
[Check this out](https://github.com/hzeller/rpi-rgb-led-matrix?tab=readme-ov-file#multiplexingl)

As I use an arrangement of 3x2 -> 32x32 Adafruit for my 96x64 Display, I no need to define a specific rgb panel, as well as is defined for default in this library. But, If I choose to use another panel with different scan rate, like outdoor 32x16, if I remember well, I MUST USE --led-multiplexing=4 so they works propperly...

[here, my screen](https://youtu.be/lurt1pDAfW0?list=PL8e1jfycOUE1-pJsEd61WE3_XFnLjYCxw)

[slowdown=2 to 4](https://github.com/hzeller/rpi-rgb-led-matrix?tab=readme-ov-file#gpio-speed), mandatory for RPi 3, and 4

davemaster, 2024-10-08:
Also... good reduce brightness for use less power, but, CHECK DOUBLE those Chinese data cable, many times those came faulty

hzeller, 2024-10-08:
Putting more than a few panels in a chain can create timing issues depending on the quality of the panels and their pass-on of the clock signal.

Don't use a single-chain connection for super-long data chains; use the parallel adapters.

AlX-Core, 2024-10-08:
With different Multiplexing settings i just get singel lines.

I checked the cable. And there are fine.

AlX-Core, 2024-10-08:
I wonder why it works with an esp32 and this lib [https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA](url)

No glitches. Same chain length.

AlX-Core, 2024-10-09:
Thanks for your fast reply :) I ordered a new board with 3 parallel. Now it works fine :)

closed 2024-10-09 by AlX-Core