hzeller/rpi-rgb-led-matrix #1717: I have no pin R2 B2 G2 E is GND GND is NC

MasterChief1335, 2024-10-12:
Hi i have no pin R2 B2 G2 E is GND GND is NC
I wanted to do this project
https://www.instructables.com/RGB-LED-Matrix-With-an-ESP8266/
[``]([image])

Thx

MasterChief1335, 2024-10-13:
???

hzeller, 2024-10-13:
If your panel don't have these pins, then you got a different kind panel, so you have to reverse-engineer how that works.

davemaster, 2024-10-14:
BUT, if You wanna use an ESP32, you're in the wrong place, this github is for use Raspberry Pi 1-4 for drive an RGB Panel... also, you miss to share your code, and circuit,  VIDEO of your output

closed 2024-10-14 by MasterChief1335

MasterChief1335, 2024-10-14:
I have now taken cable with R1G1B1 and the 3 N for RGB 3 GND connected NC I took as A3 D
[image]
[image]
[image]
[image]

Ok it's not a Pi but what difference does the pin assignment change 🤔
THX HELP

MasterChief1335, 2024-10-17:
thx no help