hzeller/rpi-rgb-led-matrix #1719: use  the p10 red led display with orange pi zero

kamu0005, 2024-10-19:
hi community,
    i am new to P10 displays. i have spents hours for the P10 display with orange pi but i got confused. can anyone tell me where to start and how. i am having orange pi zero and 32x16 P10 display(RED). i need to show some text in the display

davemaster, 2024-10-21:
The name of this github repository is rpi-rgb-led-matrix , rpi = raspberry pi, rgb =red-green-blue, IT MEANS, this library is for RGB PANELS, not  for P10 monochrome, they can be driven using arduino, and ..... library.. read a little...it hurts none

davemaster, 2024-10-21:
> Thnaks for the response. do you know any library for P10 monochrome display in python that will works on the orange pi zero

Orange pi zero, what is that, it's like a Pi zero 2W ???

Try this FIRST, you can attach and arduino for test, then, go into orange pi stuff

[Arduino P10 video with code](https://youtu.be/Z8BycdWt2Z8)

kamu0005, 2024-10-21:
i have tested the P10 with arduino it works fine. but my actual work is to use the P10 with orange pi zero.

Orange pi zero is something similar to Raspberry pi zero.

davemaster, 2024-10-22:
****

> i have tested the P10 with arduino it works fine. but my actual work is to use the P10 with orange pi zero.
> 
> Orange pi zero is something similar to Raspberry pi zero.

Well, its seems you must build your own code, a GOOD start, if the one for Arduino its doing fine, research a little deep based on that code... so you can do the same for your Orange Pi Zero board.. 

Let me help You in a couple of weeks from now, when I will have a little free time, with my Zero 2W

Best regards

marcmerlin, 2024-11-15:
I'm not seeing updates, so I'll close this for now.  Note that https://rpi-rgb-led-matrix.discourse.group/ is a better place for project help

closed 2024-11-15 by marcmerlin