hzeller/rpi-rgb-led-matrix #1725: P5 8S SM16208 panels do not work

GeorgeFlorian, 2024-10-31:
I am also encountering difficulties on HRL-P5-8S-LED1922-64x32-v2.0/P5-SMD-8S-3 that use SM16208 chips and have ABC pins.
I am using the [Active-3](https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/adapter/active-3) board.

I can't make them work at all. It's all white.
I've tried every possible combination of `--led-multiplexing`, `--led-row-addr-type` and `--led-gpio-mapping`

board707, 2024-11-22:
> What is even weirder is that the HUB75 pins are ABC and NC, and you'd think that --led-row-addr-type=3 should be the correct option

Hi
Which addressing option to use, can be easily set by the chip type on the matrix.
Please indicate the part numbers all chip on the panel other than main driver  SM16208

GeorgeFlorian, 2024-11-22:
> Hi
> Which addressing option to use, can be easily set by the chip type on the matrix.
> Please indicate the part numbers all chip on the panel other than main driver  SM16208

The black casing says P5-SMD-8S-3
The board says: HRL-P5-8S-LED1922-64x32-v2.0

There is also a code: 03N049P2E4293A2
The black chips have something like SM6208SJ written on GU,RU and BU.

On the U1 I can make up the last letters being 245B, but the first ones can be M1, MI, N1, NI. I can't really see and I don't own a microscope

[image]

board707, 2024-11-22:
Could you show a photo a whole panel ( rear side) ?

vasi26ro, 2024-12-01:
I have tried this panel with the LedArt app and Huidu D16 without success

marcmerlin, 2024-12-01:
so, I'm going to give you an honest answer, there seems to be a near infinite amount of incompatible panels, and no one is being paid to write addressing code for them, so unless you add support or personally find someone to add support for them, it's not going to happen since Henner doesn't spend his days adding support for panels he doesn't have himself, has never seen, and has no personal need for :)
If you look into the source code in the addressing/multiplexing code, you can probably figure out to add new panel types, but that's about it.
For what it's worth, normal ABC panels work with addressing=3, I have 2 batches that are 4 years apart and they both work.

board707, 2024-12-01:
If the issue is just the addressing code, then such a panel cannot be called "incompatible". Almost any type of addressing can be figured out in a few tests even without having a panel in hand.

Incompatible panels - are panels with new types of drivers, such as s-PWM drivers or drivers with a built-in buffer. Adding support for each such driver is a job that will take many weeks or even months.

marcmerlin, 2024-12-01:
@board707  we can argue on semantics, but the point remains. Unsupported panels will remain unsupported until someone provides code to support them. If there is an expectation that this will come from Henner, I know him personally have talked with him recently, and he has plenty of other things to do.
That said, you seem to know what's going on, so please feel free to step in and provide pull requests, those will get looked at.

board707, 2024-12-01:
> That said, you seem to know what's going on, so please feel free to step in and provide pull requests

To do that, I need to get a Raspberry Pi board in my hands :) Up to now, I used HUB75 screens with low memory controllers only - STM32, RP2040 etc.
I am planning to test your library with Pi Zero.

marcmerlin, 2024-12-01:
get a Pi0 2W, it's cheap and almost as capable as a rPi3. I just benchmarked it last week:
https://rpi-rgb-led-matrix.discourse.group/t/rpi0-2w-speed-compared-to-rpi3a/913/2

board707, 2024-12-02:
Thanks for the link, I just registered there.

GeorgeFlorian, 2024-12-03:
Oh my God, this got a lot of traction and attention.

I understand that it's impossible to get every panel working, I just hopped to get some pointers as to what I can try to get this one working. The supplier that I get my panels from only has these ones now, which I cannot make work.

It is obviously an addressing issue. I have tried every possible combination of settings and none outputted a correct image.

Unfortunately, I am not that good of a programmer to figure out the addressing.
I am even willing to pay somebody to help me with these panels as these are the only ones I can get my hand on right now.

ScobbyDoo, 2024-12-03:
Same issue with me , I've made simple solution.

I've managed to get same lot of led module which is supported by setting i tried.

Instead of Re Mapping leds & put efforts on this kind of solution, batter to switch on supported modules , it's helpful for other library too.

vasi26ro, 2024-12-03:
@GeorgeFlorian I am working with a similar panel here in Romania, I have send you a friend request in linkedin, maybe we can find a solution.

By now I have found a way to configure correctly the panel with a huidu hd d16 controller and I captured the signals with a logic analyzer. 

I would try to replicate this signals from this library. Hopefully I can write the mapper. 

With this command: `sudo ./pixel-mover  --led-row-addr-type=0 --led-multiplexing=19 --led-multiplexing=19 --led-slowdown-gpio=2 --led-rows=32 --led-cols=64` the bottom half of the panel works. The first half is always white.

board707, 2024-12-03:
@GeorgeFlorian , @vasi26ro , @ScobbyDoo 
Colleges, figuring out a pixel mapping is not as difficult as it seems. I have developed a method that allows me to configure a pixel mapping for almost any panel, and for most panels I do not need to have it in hand - it is enough for you to run several tests and show the result on video.

The problem is that I do not work with the hzeller library. I have my own DMD_STM32 library and it is written for boards on the STM32 and RP2040 controllers. In issue #1686, one of the users already referred to my library as an example of working with complex pixel mapping.

For those who think it is possible to use other controllers - I suggest testing your panels on my library. In the future, I plan to try to add some mapping options to this library, but this will take time, since I have never worked before with Raspberry boards

ScobbyDoo, 2024-12-04:
So , it means whatever buffer any library is generating, 
Pass that buffer to remapping function & cast pointer at same position but just circular buffer.

It means same buffer will rearrange, with any other same size static buffer , basically its just lookup table which works as in pair of 2 4 8 or 16 bytes & skipping positions.

So what portion of your library's snippet will help to remap any led module in that case? I think chatgpt helps for quick implementation

board707, 2024-12-04:
I'm not sure I understood exactly what you mean.
The general principles of pixel mapping are the same, but the specific implementation depends on the library. I am not using a circular buffer.

> what portion of your library's snippet will help to remap any led module

What I wrote does not mean that I already have code for any matrix, I only developed the principles of writing it for a specific panel. The main code for working with pixel mapping in my library is located in the file 
https://github.com/board707/DMD_STM32/blob/dev-V2/DMD_Panel_Templates.h

The method of figuring out of the mapping is described in another library discussion -
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/discussions/622

vasi26ro, 2024-12-05:
I have tried the following command `sudo ./pixel-mover  --led-row-addr-type=0 --led-slowdown-gpio=2 --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat --led-multiplexing=19 --led-brightness=100 --led-limit-refresh=100 -t 10`

This looks almost perfect, but there is some ghosting going on. If you can point me where in code to look maybe I can find a solution faster.

I am using this code: https://github.com/hzeller/rpi-rgb-led-matrix/compare/master...briggsm:rpi-rgb-led-matrix:p5outdoor8s64x32MultiplexMapper

https://github.com/user-attachments/assets/356c271e-f9c6-4f65-a75c-d111e2c80299

board707, 2024-12-05:
> This looks almost perfect,

Huh?
What was supposed to be shown on the screen?

To test the mapper I would recommend to run a code that fulfill a panel pixel by pixel from upper left to lower right corner.

vasi26ro, 2024-12-05:
In the video, i am using pixel-mover from the `examples-api-use/` dir.
This example allows you to move a pixel around the screen from (0,0) top left, anywhere on the screen. Also the argument -t 10 adds another 10 pixels as tail to the leading pixel.

By moving this pixel, I can see that the mapping is correct, but there is some ghosting and flickering. If it helps, I can get capture a part of the signals from the hub75. I also have a capture with the working signals from the huidu led controller.

vasi26ro, 2024-12-05:
The setup that I have:

- rpi 4b
- Adafruit RTC HAT
- The hat is in quality mode with GPIO18 soldered to GPIO4

Here is the rotating square from: `sudo ./demo -D0  --led-row-addr-type=0 --led-slowdown-gpio=2 --led-rows=32 --led-cols=64 --led-gpio-mapping=adafruit-hat --led-multiplexing=19 --led-brightness=100 --led-limit-refresh=100`

https://github.com/user-attachments/assets/7c6afda5-f78e-4ff3-8ee4-2449d5ae46a3

board707, 2024-12-05:
Sorry, this video show nothing for me about mapping.

Could you run something this:
```
 canvas->Fill(0, 0, 0);
 for (int y =0; y < canvas->height(); y++) {
   for (int x =0; x < canvas->width(); x++) {
    canvas->SetPixel(x, y, 255, 0, 0);
    usleep(30 * 1000L); // wait a little to slow down things.
  }
  }
```
and show on the video about 10-15 seconds ?

vasi26ro, 2024-12-05:
https://github.com/user-attachments/assets/3a5ddd87-99b6-4781-a458-5ca18da14700

board707, 2024-12-05:
Thanks
Looks like a problem with the row switch. 
Remind me what panel you have (dimensions, scans) and what driver chips are installed on it.

vasi26ro, 2024-12-06:
The panel is the same one as @GeorgeFlorian `HRL-P5-8S-LED1922-64x32-V2.0` with SM16208SJ MW245B, others I do not have noted and I am away from the panels until Sunday or Monday.

[image]
[image]
[image]

board707, 2024-12-06:
@vasi26ro 
Could you show a photo a whole panel ( rear side) ?

@GeorgeFlorian  - the same question for you

vasi26ro, 2024-12-06:
[image]

This is the best image that I have.

board707, 2024-12-22:
@GeorgeFlorian , @vasi26ro , @ScobbyDoo
Lets continue our conversation about pixel mapping.
Now I received a Pi Zero. I just tested hzeller's library in real hardware and successfully tried to figure out a pixel mapping on few my panels.

If the problem of choosing correct mapping for your panels is still actual, we could try to solve it together. All that is required from you is to run several tests with the code that I will send and show the result on video.

@vasi26ro 
Last time I made a mistake testing your matrix, so I decided that it has problems with the driver. Maybe it's not that bad :)

marcmerlin, 2024-12-23:
@board707  figuring out the mapping is not exactly hard, and those panels may be zag zig which is already known as a mapping.
The missing work is to write an alternate mapper in the code. Are you offering to write one in a way that it's reasonable to merge in the main codebase?

vasi26ro, 2024-12-23:
@board707
Thank you, can you tell me what tests to run?

marcmerlin, 2024-12-23:
> @board707 Thank you, can you tell me what tests to run?

you can simply displays dots one by one and see where they end up, but as I wrote, the mapping is likely already known, what's missing is for you or him or someone to write the actual code for the mapping. 
Until someone volunteers to do that, not much will happen.

board707, 2024-12-23:
> Are you offering to write one in a way that it's reasonable to merge in the main codebase?

Yes, that's exactly what i'm going to do. 
After i got used to the library, i wrote mappers for two of my panels yesterday.

board707, 2024-12-23:
> can you tell me what tests to run?

please run this code again:
```
canvas->Fill(0, 0, 0);
 for (int y =0; y < canvas->height(); y++) {
   for (int x =0; x < canvas->width(); x++) {
    canvas->SetPixel(x, y, 255, 0, 0);
    usleep(30 * 1000L); // wait a little to slow down things.
  }
  }
```

but now with settings `--led-rows=16 --led-cols=128`. 
Do not set any special mapper.

board707, 2025-01-13:
> Until someone volunteers to do that, not much will happen.

But as you can see, those who ask for help - do not go beyond complaints.

board707, 2025-01-13:
@vasi26ro 
I fully understand that everyone has their own concerns, especially since there are holidays and Christmas :)
But it would be better to warn me that you will not be able to do this right away. For this setup, we may need to perform about a dozen tests - and it is advisable to do it over several days, so as not to forget what we were going to do :)

vasi26ro, 2025-01-15:
@board707 

Here is the video with `--led-rows=16 --led-cols=128` and no special mapper

https://github.com/user-attachments/assets/987ee3c8-bb32-44ab-bfdb-b590fdd46570

board707, 2025-01-15:
Thanks.
I should analyze it.

board707, 2025-01-16:
Ok, lets start :)

Add a new mapper to the `lib/multiplex-mappers.cc` file:
```
class vasi26ro_Mapper : public MultiplexMapperBase {
public:
  vasi26ro_Mapper() : MultiplexMapperBase("vasi26ro_Mapper", 2) {}

  void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const {
     const bool is_top_stripe = (y % (panel_rows_/2)) < panel_rows_/4;
     const int pixelbase = 4;
    *matrix_x = (x/pixelbase)*pixelbase*2;
    if (is_top_stripe) *matrix_x += pixelbase + (x % pixelbase);
    else *matrix_x += x % pixelbase;
    *matrix_y = ((y / (panel_rows_/2)) * (panel_rows_/4)
                 + y % (panel_rows_/4));

 }
}; 
```
Do not forget to register the new mapper in mapper list table, adding a new line to the `CreateMultiplexMapperList()` function:

```
result->push_back(new vasi26ro_Mapper()); 
```

Than rebuild the library and run it with a mapper number that is obviously greater than the number of mappers in the table to get a list of all available mappers.
Find our new mapper in the list and run your test code with it.

vasi26ro, 2025-01-16:
https://github.com/user-attachments/assets/1a739278-2634-4e2d-bcfe-a85288f24ca7

Woow @board707 it looks way better but there is this ghosting. I am using adafruit-hat-pwm with rpi4. The cables are ok, if I use this with another controller it works.

board707, 2025-01-16:
> there is this ghosting.

It could be a problem with contacts (probably A B C lines) or with the multiplexer chip.
As far as I remember, in December we still couldn't figure out what chip you had under the plastic frame (see the photo from December 6 above). Without knowing the type of chip, it's impossible to say whether it works correctly.

But in any case, all this has nothing to do with the pixel mapping -  we solved that problem.

vasi26ro, 2025-01-17:
In the name of the science, I have cut the frame with a dremel (and a piece of the pcb :)) by mistake), the driver under the frame is Chipone ICN 2012. I cannot make a photo. @board707

board707, 2025-01-17:
Wow!!
On one of my panels I did this using a children's device for burning patterns on wood - its hot spiral cuts plastic perfectly. As a side benefit, there is less chance of damaging the panel.

Unfortunately, we did not see anything particularly new under the plastic - ICN2012 is a standard multiplexer, for which the option `--led-row-addr-type=0` you selected is perfectly legal. So the ghosting problem is not there.

board707, 2025-01-17:
@vasi26ro 
Do you using a `--led-no-hardware-pulse` option?
Also see a Troubleshooting  section in the Readme.
 
And next - when you tried to run this panel with Huidu D16 - did you seen the same ghosting?

GeorgeFlorian, 2025-03-20:
@vasi26ro @board707 
Thank you for your interest in this issue. I am sorry it took me so much time to respond, but I was caught up in other projects.

I can confirm that @board707 mapper works with my panels:
```cpp
    class P5_8S_64x32 : public MultiplexMapperBase
    {
    public:
      P5_8S_64x32() : MultiplexMapperBase("P5_8S_64x32", 2) {}

      void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const
      {
        const bool is_top_stripe = (y % (panel_rows_ / 2)) < panel_rows_ / 4;
        const int pixelbase = 4;
        *matrix_x = (x / pixelbase) * pixelbase * 2;
        if (is_top_stripe)
          *matrix_x += pixelbase + (x % pixelbase);
        else
          *matrix_x += x % pixelbase;
        *matrix_y = ((y / (panel_rows_ / 2)) * (panel_rows_ / 4) + y % (panel_rows_ / 4));
      }
    };
```
And I have also found another mapper that works with these panels: https://github.com/hzeller/rpi-rgb-led-matrix/pull/1544
```cpp
    class P5Outdoor8S64x32MultiplexMapper : public MultiplexMapperBase
    {
    public:
      P5Outdoor8S64x32MultiplexMapper() : MultiplexMapperBase("P5Outdoor8S64x32", 2) {}

      void MapSinglePanel(int x, int y, int *matrix_x, int *matrix_y) const
      {
        int pix_nb = x + (y * 64);
        *matrix_x = ((((pix_nb % 64) / 4) + 1) * 8) - ((((pix_nb % 1024) / 512) + 1) * 4) + (pix_nb % 4);
        *matrix_y = ((pix_nb / 64) % 8) + ((pix_nb / 1024) * 8);
      }
    };
```

Also, to improve stability and to remove flicker and ghosting on RPI 4, I have modified the upper limit of gpio_slowdown up to 10:
https://github.com/hzeller/rpi-rgb-led-matrix/blob/d10f518f291bd34ca4816ef25cadf2446b882662/lib/led-matrix.cc#L661
```cpp
  if (runtime_options.gpio_slowdown < (LED_MATRIX_ALLOW_BARRIER_DELAY ? -1 : 0)
      || runtime_options.gpio_slowdown > 10) {
    fprintf(stderr, "--led-slowdown-gpio=%d is outside usable range\n",
            runtime_options.gpio_slowdown);
    return NULL;
  }
```

I am using `--led-slowdown-gpio=6` and there is no flicker in any of the simple demos I have tried.
I have not rendered an image or a video, because it is not in the scope of my project.

board707, 2025-03-20:
> I can confirm that [@board707](https://github.com/board707) mapper works with my panels:

Thanks for your feedback.

I can try to help in pixel mapping if someone has problems with it. 
But it is necessary that the person is ready to make several tests with his matrix, and it is desirable that it does not take several months :)

marcmerlin, 2025-03-24:
@GeorgeFlorian  what are the odds, I hit a similar problem with ABC panels that did not work without loads of noise until I changed the slowdown to 8, which was beyond allowable values.
I did just submit https://github.com/marcmerlin/rpi-rgb-led-matrix/commit/03c209c9eadee4e76086180b3b98813eec86a447
which now allows this, but it's just papering over the real problem.

Questions
1) are you using  --led-row-addr-type=0 or  --led-row-addr-type=3 like I had to? (in bug  https://github.com/hzeller/rpi-rgb-led-matrix/issues/1774 )
2) please read https://github.com/hzeller/rpi-rgb-led-matrix/issues/1773#issuecomment-2746672857 , my output got better with a much older raspbian 4 32bit compared to a newer dietpi 6 64bit. This may help you as a workaround (not a fix)

After that, I'm not equipped with a proper scope, so I can't debug the code and waveforms. I did mention this to @hzeller although he just left for a trip, so he won't be able to look for a while.
@board707  I appreciate your willingness to help and contributions recently.  Have you been able to get yourself rPi hardware recently and panels to replicate some of those things locally?  Are you equipped with a scope to look at waveforms and help debug?

marcmerlin, 2025-03-24:
Oh, now I see your first picture showing it is an ABC panel. 
Given that you also had to raise the gpio slowdown, I'm going to assume you now have the same issue as I do, so I'm going to dupe your bug against mine.  Please feel free to answer there.

marcmerlin, 2025-03-24:
@GeorgeFlorian please followup on the new bug with my questions 1 and 2 above, thanks.
and at this point, we'll need someone to dig into the SM16208 spec sheet and compare it with ICN2037
that may be enough to figure out why they are kind of compatible, but not fully.

board707, 2025-03-24:
> [@board707](https://github.com/board707)  Have you been able to get yourself rPi hardware recently and panels to replicate some of those things locally? Are you equipped with a scope to look at waveforms and help debug? 

To work with panels I use a 100 MHz logic analyzer, which in my opinion is much more convenient than an oscilloscope.
I will try to help as much as I can, but unfortunately my experience is with microcontrollers such as the STM32. I have never worked with a rPi and know nothing about the library's internals.

GeorgeFlorian, 2025-03-26:
> [@GeorgeFlorian](https://github.com/GeorgeFlorian) please followup on the new bug with my questions 1 and 2 above, thanks. and at this point, we'll need someone to dig into the SM16208 spec sheet [...]

I have had issues with the dietpi and also had to focus on other projects these last days. I will test things today and will answer your questions.

marcmerlin, 2025-03-26:
@board707  I agree a logic analyzer is much better if you have one :)
I've been a bit puzzled by your help without actually using the library. Clearly, you understand how panels work, as well as the logic and driving requirements.  You are also better equipped than me  :)
If you work on STM32, you also must know C/C++, am I correct?
If so, can I invite you to buy a cheap rpi3a or rpi0 2 wireless, for maybe $30 or so, and try the library?   It should be trivial for you to get an rPi to boot since lost of people less capable than you are doing it daily ;)  and if you can read STM32 code, reading the C++ (but it's mostly C) code for this library should be simple too. I'm far from being a star programmer myself, but  I was able to quickly understand Henner's code where needed for small changes I needed.

marcmerlin, 2025-03-26:
@GeorgeFlorian wireless worked fine for me on dietpi with the configurator, but in this case, you might try an older 32bit raspbian and see if it reduces your need for a big GPIO slowdown, like it did for me.
It's not a fix, but it's a nice workaround in the meantime.

board707, 2025-03-26:
> can I invite you to buy a cheap rpi3a or rpi0 2 wireless, and try the library?

@marcmerlin, I probably didn't make myself clear - I have an RPI Zero and I ran the library code on it. I used this setup to test the multiplex mapper code variations I suggested earlier in the thread.

marcmerlin, 2025-03-27:
Tnanks @GeorgeFlorian .  Rpi4 does work faster and better than Pi3 (which i only use because it's smaller and I'm running on batteries), so you're good with what you have.
The flicker is annoying, try isolcpus=3 as explained in the docs

marcmerlin, 2025-03-27:
> [@marcmerlin](https://github.com/marcmerlin), I probably didn't make myself clear - I have an RPI Zero and I ran the library code on it. I used this setup to test the multiplex mapper code [...]

Understood, thanks for that and helping out.
If it is available where you live, try to get a Pi 0 2 wireless which is about as fast as a Pi3a but much smaller.
This will give you full speed and performance, as well as 3 channel output while the original Pi0 is pretty limited in output and speed

GeorgeFlorian, 2025-03-27:
> Tnanks [@GeorgeFlorian](https://github.com/GeorgeFlorian) . Rpi4 does work faster and better than Pi3 (which i only use because it's smaller and I'm running on batteries), so you're good with what you have. The flicker is annoying, try isolcpus=3 as explained in the docs

I have followed every instructions in the readme to the letter:
- minimal DietPi;
- tried both hardware pulse enabled and disabled (enabled had better results, less flicker)
- `dtparam=audio=off` in `/boot/firmware/config.txt`;
- no `dtoverlay=w1-gpio` in `/boot/firmware/config.txt`
- ran `sudo apt-get remove bluez bluez-firmware pi-bluetooth triggerhappy pigpio`; No extra package installed
- we have custom made adapters: https://github.com/hzeller/rpi-rgb-led-matrix/tree/master/adapter/active-3
- `isolcpus=3` at the end of `boot/firmware/cmdline.txt`; No new line.

All that really improved flicker was bigger gpio_slowdown values no matter the panel or the OS used.

board707, 2025-03-27:
@marcmerlin 
In what file the code, related to --led-row-addr-type=3 option is located?

marcmerlin, 2025-04-02:
Sorry for the late reply @board707  sorry been very busy travelling (back to back trips). Surely you can grep the code? :)
if you can't find it and you need me to do it for you, let me know and I'll look when I can.

board707, 2025-04-02:
@marcmerlin 
of course it's not hard for me to find the right piece of code in the library. I asked only in case there are some complex interactions between parts of the project that are worth paying attention to.

So, the row multiplexer is in the `framebuffer.cc` file. 
There are 5 types of multiplexers in total, in the order of the `--led-row-addr-type` option value:

0. DirectRowAddressSetter()
1. ShiftRegisterRowAddressSetter()
2. DirectABCDLineRowAddressSetter()
3. ABCShiftRegisterRowAddressSetter()
4. SM5266RowAddressSetter()

As far as I understood the code, **none of these multiplexers are compatible with the "type 595 shift_register" such as SM5368 or DP32020 chips.**

Judging by the name, the `--led-row-addr-type=3` option is seems the best fit. But in fact this multiplexer is not ABCShiftRegister. The code uses only pins A and C.

The correct code for the ABCmultiplexer should be the same as for a regular shift register, with three signals - clock,  data and latch. The closest to such a code is setting the lower 8 bits in the SM5266RowAddressSetter() option.

In addition, as far as I can see, in all multiplexers derived from the shift register, you each time push into the multiplexer the number of bits corresponding to the full number of lines. In fact, it is enough to push one bit into the multiplexer when changing a line - it will be faster and there is less chance of glitches on the panel.

marcmerlin, 2025-04-02:
1) @board707  I agreed that ABC panels are really AC panels like you said.
2) I didn't write any of those code, I just filed the original bug that new ABC panels existed and didn't work
3) I think Henner wrote most of the exist code for AC panels
4) we also agree that the code may have something not quite right in it (I honestly haven't had any time to look, and won't for maybe 1-2 months due to back to back travel)

If you have an ABC panel of any kind and you feel inspired to fix the code to properly support those shift registers, it will sure be appreciated by multiple users, including me.

If so, can I ask you to followup on https://github.com/hzeller/rpi-rgb-led-matrix/issues/1773, which is now the master bug for tis issue?

board707, 2025-04-02:
> can I ask you to followup on [#1773](https://github.com/hzeller/rpi-rgb-led-matrix/issues/1773), which is now the master bug for its issue?

Sorry, I don't understand you, where I should send my messages... If the thread #1773 is a master bug, why did you closed it?

[2 more replies]