hzeller/rpi-rgb-led-matrix #1726: Issue with temperature sensor connected to GPIO 26

wailla, 2024-11-02:
Good morning, according to the documentation pins 29 to 40 are used only not in the case of a single panel and that is why I would like to use it to read a sensor, but it keeps giving me error :

fabio@raspberrypi:~/project/projectPanelLed $ sudo python test3.py 
Traceback (most recent call last):
  File "/home/fabio/project/projectPanelLed/test3.py", line 21, in <module>
  File "/usr/local/lib/python3.11/dist-packages/Adafruit_DHT/common.py", line 81, in read
    return platform.read(sensor, pin)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/dist-packages/Adafruit_DHT/Raspberry_Pi_2.py", line 34, in read
    raise RuntimeError('Error accessing GPIO.')
RuntimeError: Error accessing GPIO.

I ask why I can't read sensor if the sensor test code is this.

import time
from rgbmatrix import RGBMatrix, RGBMatrixOptions
import Adafruit_DHT

options = RGBMatrixOptions()
options.rows = 64
options.cols = 64
options.chain_length = 1 # only one panel
options.parallel = 1
options.hardware_mapping = 'regular'
matrix = RGBMatrix(options=options)

DHT_SENSOR = Adafruit_DHT.DHT11
DHT_PIN = 25 # Use an unused pin if there are conflicts

while True:
    humidity, temperature = Adafruit_DHT.read(DHT_SENSOR, DHT_PIN)
    
    if humidity is not None and temperature is not None:
        print(f'Temperature: {temperature:.1f}°C, Humidity: {humidity:.1f}%')
    else:
        print(“Error reading the sensor.”)
    
    time.sleep(2) 

PS: the sensor works since with other scripts the required output comes out.

poet-of-the-fall, 2025-04-07:
I'm also using an Adafruit Gyro Sensor (https://github.com/adafruit/Adafruit_CircuitPython_MPU6050/tree/main?tab=readme-ov-file) and also fail to get both working at the same time. I woulds also be interested in suggestion how to get both working at once.

marcmerlin, 2025-11-27:
With apologies to everyone, this project is doing a one time cleanup of 
all the bugs that have not been addressed and closed in the last 10 years.

If you filed an issue to ask for help, please do not re-open this one or a new
bug, but go here instead: https://rpi-rgb-led-matrix.discourse.group/

Please be mindful that this project does not have any full time maintainers
or anyone paid. You should not expect personal help for your own projects or
answers to programming questions but you can ask for help with panels that
aren't working, setup questions, or anything that may be a bug here:
https://rpi-rgb-led-matrix.discourse.group/
If you need programming help and are willing to pay someone for it, please
state so in your message, and someone may be interested.

If you did report an actual bug that is still not fixed today, please accept
our apologies for this being mass closed, you can absolutely send a followup
and re-open the issue, or open a new up to date one as applicable.
Note that we very much welcome help, so if you can help fix your bug, you will
be a superstar :)

If your bug was linked to a pull request that was not resolved, please update
the pull request and the issue, and we will do our best to review them. We do
thank you for your undertanding that we have very limited free time, so any
extra work you do along with your bug report or PR request will be very much
appreciated and improve the odds that they will be looked at.

If you want to help contribute as a python or C# or any other language bindings,
you are also a superstar, please state so in the issue and we'll do our best to
give you the access and reviews you need.

For anything related to PWM panel support, please make sure you read README and
then refer to https://github.com/hzeller/rpi-rgb-led-matrix/issues/466 . 
Please do not comment if you're haven't read everything and/or if you're just
asking when your panel is going to be supported. Those bugs are if you're 
able to contribute to the effort.

For rPI5: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1603

For anything python, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828
For anything C#, please link your bug to: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1828