hzeller/rpi-rgb-led-matrix #1729: AZERONE 128x64 ABC FM6124 Led Matrix not working

jacksonhmg, 2024-11-06:
Hey guys,

I bought this panel:
https://www.amazon.com/gp/product/B0869P1DCH

It is: 
- 128x64
- ABC addressing (even though the description says ABCDE)
- Uses chips FM6124 and 74HC245C
- P2

[image]
[image]

And I cannot get it working with any configuration.

I noticed this similar issue:
https://github.com/hzeller/rpi-rgb-led-matrix/issues/823

and this:
https://github.com/hzeller/rpi-rgb-led-matrix/issues/1025

However neither of suggestions in them seem to work.

For D0:

All --led-row-addr-type=0 variations all produce some variation of this result:
[image]

All --led-row-addr-type=1 variations all produce some variation of this result:
[image]

All --led-row-addr-type=3 variations all produce some variation of this result:
[image]

Experimenting with multiplexing and chip settings has not given any good results.

Do these results signal anything to those more experienced?

(@marcmerlin i noticed you used a similar board, is there anything that stands out about my output? maybe a signal of poor powering?)

Any help would be really appreciated, thank you.

marcmerlin, 2024-11-07:
Ok, I had a look at my ABC and ABCDE panels from Azerone, and indeed you have the ABC.
First thing is I would contact azerone and ask for a replacement. The amazon listing says ABCDE and you got the wrong panel so they should send you the ones you paid for.
I'm sure you saw https://github.com/hzeller/rpi-rgb-led-matrix/issues/823 I posted in 2019, and strangely you got the same error than I did then, I contacted azerone and they sent me the correct panel. 
It's probably best if you go through amazon support, but if needed, you can contact https://www.facebook.com/su.hongren.9 from Azerone directly.

Now, as per the link I posted, those ABC panels, can work, mine did, but I chose not to work them since I was setup to use ABCDE and wanted all my panels to be compatible.

As per the other bug I linked, for me the ABCs did work with
~/rpi-rgb-led-matrix/examples-api-use/demo --led-rows=64 --led-cols=128 --led-chain=1 --led-show-refresh --led-pwm-dither-bits=1 --led-pwm-bits=7 --led-panel-type=FM6126A --led-row-addr-type=3 -D0 
but I still recommend you return them.

marcmerlin, 2024-11-07:
@hzeller  as per our discussion last weekend, I don't know how much I can promise to help with bug-wise, but can you either dupe this bug against #823 , or give me github bug admin access and I'll be happy to do so?

marcmerlin, 2024-11-15:
this is a dupe of the bug I linked, closing this

closed 2024-11-15 by marcmerlin