hzeller/rpi-rgb-led-matrix #1732: Custom mapping

barryVDs, 2024-11-13:
Hi All,

I am building a digital analog clock, with a creative layout using Waveshare LED matrix panels 64X32.

[image]

[image]

Can I build a custom mapper, to get this working? I think the biggest issue is the 2nd and 5th row, because these are not aligned with the rest.

Also, is it possible to rotate the screen 45 deg? Or should I do this in the content?

If I do a per pixel mapping, I am afraid this will overload the processor (I do want to do some small background animations)

I have a RPI4 headless, 32 bit (can install 64bit if this is better, installed 32bitt for compatibility)
Adafruit hub75 board
12X waveshare 64X32 Led Matrix

board707, 2025-05-22:
@barryVDs 
It's very interesting, could you sharing the code?

Do you think that orienting screens with a tilt was the right decision? After all, most raster fonts are designed for horizontal arrangement of rows of pixels, in such a position as yours, output to the screen is complicated many times over

barryVDs, 2025-05-22:
[image]
[image]

Hi,

I am very happy with the result, there can be color shift if you rotate a LED matrix (I do work with LED matrixes professionally) but on this project it is not an issue.
The extra handling is ok to do, I just rotate the output 45deg using PIL.

This is my code:

import json
import os
import math
from datetime import datetime
from PIL import Image, ImageDraw, ImageFont
import pytz

CONFIG_PATH = "/mnt/nfs_clientshare/settings/klok.json"

def load_config(path=CONFIG_PATH):
    if not os.path.exists(path):
        raise FileNotFoundError(f"Configbestand niet gevonden: {path}")
    try:
        with open(path, "r") as f:
            return json.load(f)
    except json.JSONDecodeError as e:
        raise ValueError(f"Fout in JSON-formaat: {e}")

def rgba_from_config(cfg, prefix, default=(255, 255, 255, 255)):
    try:
        return (
            cfg[f"{prefix}_r"]["value"],
            cfg[f"{prefix}_g"]["value"],
            cfg[f"{prefix}_b"]["value"],
            cfg.get(f"{prefix}_a", {"value": 255})["value"]
        )
    except KeyError:
        return default

def draw_analog_clock(background):
    config = load_config()

    # Basisconfiguratie
    center_x = background.size[0] // 2
    center_y = background.size[1] // 2
    mask_radius_factor = config.get("mask_radius_factor", {"value": 1.0})["value"]
    clock_radius = int(min(background.size) // 2 * mask_radius_factor)
    marker_distance_text = config.get("marker_distance_text", {"value": 0.85})["value"]
    marker_distance_dots = config.get("marker_distance_dots", {"value": 0.85})["value"]
    marker_distance_quarters = config.get("marker_distance_quarters", {"value": marker_distance_text})["value"]

    hour_length = config.get("hour_hand_length", {"value": 50})["value"]
    minute_length = config.get("minute_hand_length", {"value": 70})["value"]
    hand_width = config.get("hand_thickness", {"value": 4})["value"]

    hour_color = rgba_from_config(config, "hand_color")
    center_color = rgba_from_config(config, "center_point_color")
    fill_color = rgba_from_config(config, "mask_color")
    border_color = rgba_from_config(config, "background_color")

    center_size = config.get("center_point_size", {"value": 2})["value"]
    marker_type = config.get("marker_type", "dots")
    marker_radius = config.get("marker_radius", {"value": 12})["value"]
    font_size = config.get("font_size", {"value": 14})["value"]

    marker_text_color = rgba_from_config(config, "marker_text_color", (0, 0, 0, 255))
    marker_background = rgba_from_config(config, "marker_background", (255, 255, 255, 255))

    # Romeinse cijfers
    roman_numerals = ["XII", "I", "II", "III", "IV", "V", "VI", "VII", "VIII", "IX", "X", "XI"]
[... 110 lines cut ...]
            if marker_type == "numbers_in_circles":
                blur_amount = config.get("marker_blur", {"value": 2})["value"]
                soft_circle = Image.new("RGBA", background.size)
                circle_draw = ImageDraw.Draw(soft_circle)
                circle_draw.ellipse(
                    (x - marker_radius, y - marker_radius, x + marker_radius, y + marker_radius),
                    fill=marker_background
                )
                from PIL import ImageFilter
                blurred = soft_circle.filter(ImageFilter.GaussianBlur(blur_amount))
                clock_image = Image.alpha_composite(clock_image, blurred)
                draw = ImageDraw.Draw(clock_image)

            text_size = draw.textsize(text, font=font)
            text_x = x - text_size[0] / 2
            text_y = y - text_size[1] / 2
            draw.text((text_x, text_y), text, fill=marker_text_color, font=font)

    # Middelpunt
    draw.ellipse(
        (center_x - center_size, center_y - center_size, center_x + center_size, center_y + center_size),
        fill=center_color
    )

    return Image.alpha_composite(background.convert("RGBA"), clock_image).convert("RGB")

board707, 2025-05-22:
Thank you. It's very interesting project