hzeller/rpi-rgb-led-matrix #1733: RFE/Placeholder: Support for 6 port board: Beaglebone Pocketcroller V2

marcmerlin, 2024-11-15:
This board: https://kulplights.com/product/pocketscroller/
[image]

This is beaglebone so it already runs linux, I have no idea if they took this code and ported it to that board, or if they wrote their own code.  The project points to https://github.com/FalconChristmas/fpp/
Maybe the project contributors can give us a quick clue on which code is being used, and whether people using rpi-rgb-panel should move to their driver if they need more than 3 channels (up to 6) or whether it actually is a port of this code and things will "just work"

darylc, 2025-01-11:
The code there uses Beaglebone specific PRU hardware that is not present on rpi

marcmerlin, 2025-01-12:
@darylc  understood that BB has different (better) IO options.  The question I meant to ask was "if someone needs more than 3 channels of output and is using the rPI driver", is your code sufficiently lib-like and similar that people can buy a BB, retarget their code against your driver ideally with minimal work, and then be able to have 6 channel output?
Given that both boards run linux with gcc, the only issue should be about the library API itself, am I correct?

marcmerlin, 2025-01-13:
@darylc replied on the other repo that the display code there is not meant to be used as a standalone library.
Of course, for someone sufficiently motivated, the code could still be used as a display engine for code written against rpi-rgb-panel, but it would be non trivial work, since it wasn't designed for that.

closed 2025-01-13 by marcmerlin