hzeller/rpi-rgb-led-matrix #1738: update color based on a number

dnielso5, 2024-12-10:
Hello, part of a code i run looks at the temperature and outputs it to the matrix in a solid color, i know some basic coding but not something like this. how would it be updated to change color based on the temp?

            # draw the min/max temperatures
            graphics.DrawText(
                self.offscreen_canvas,
                self.fontMed,
                50 + 36,
                3 * lineHeight + self.lineSpacing,
                graphics.Color(0, 255, 128),
                str(int(minTemp)) + "/" + str(int(maxTemp)) + u"\u00b0" + "F",

and

            # draw the current temperature
            graphics.DrawText(
                self.offscreen_canvas,
                self.fontMed,
                2,
                3 * lineHeight + self.lineSpacing,
                graphics.Color(0, 255, 0),
                str(int(self.temperatureNow)) + u"\u00b0" + "F",

dnielso5, 2024-12-11:
I thought this would work but i get an error saying the if is invalid syntax

```
                # draw the current temperature
                graphics.DrawText(
                    self.offscreen_canvas,
                    self.fontMed,
                    2,
                    3 * lineHeight + self.lineSpacing,
                    graphics.Color(
		            if int(self.temperatureNow) < 40:
			           graphics.Color(0, 0, 255)
		            else: 
			           graphics.Color(0, 255, 0)
                    ),
                    str(int(self.temperatureNow)) + u"\u00b0" + "F",
```

dnielso5, 2024-12-11:
found the way to do it, needs some adjustment but here is the basics

```
                # Determine the color based on the temperature
                if int(self.temperatureNow) < 40:
                   color = graphics.Color(0, 0, 255)  # Blue for cold temperatures
                else:
                   color = graphics.Color(0, 255, 0)  # Green for warmer temperatures
                # draw the current temperature
                graphics.DrawText(
                   self.offscreen_canvas,
                   self.fontMed,
                   2,
                   3 * lineHeight + self.lineSpacing,
                   color,
                   str(int(self.temperatureNow)) + u"\u00b0" + "F",
                )
```

dnielso5, 2024-12-15:
```
                def get_temperature_color(temp, min_temp, max_temp):
                # Map the temperature to a color in the range -20°F to 120°F, with yellow in the middle.
    
                    if temp <= -20:
                        return graphics.Color(0, 0, 255)  # Blue for cold
                    elif temp >= 120:
                        return graphics.Color(255, 0, 0)  # Red for hot
                    elif temp <= 70:
                        # From -20°F to 60°F, color transitions from blue to green
                        normalized_temp = (temp + 20) / 90  # Normalize between 0 and 1 for -20°F to 60°F
                        red = 0
                        green = int(255 * normalized_temp)
                        blue = int(255 * (1 - normalized_temp))
                        return graphics.Color(red, green, blue)
                    else:
                        # From 70°F to 120°F, color transitions from green to red
                        normalized_temp = (temp - 70) / 50  # Normalize between 0 and 1 for 60°F to 120°F
                        red = int(255 * normalized_temp)
                        green = int(255 * (1 - normalized_temp))
                        blue = 0
                        return graphics.Color(red, green, blue)
```

closed 2024-12-15 by dnielso5

marcmerlin, 2024-12-18:
for future, please use https://rpi-rgb-led-matrix.discourse.group/