hzeller/rpi-rgb-led-matrix #1739: Panel refresh limitation

nrichmon, 2024-12-11:
I am trying to understand how it's possible to get such high refresh rates as some people seem to. Here a trivial example
32 x 32 panel, 1:16 scan, 8bpp. Panel has max refresh of 30Mhz. I assume this applies to the rate at which bits can be shifted in. 

Bits required per row = 32 x 8 (cols x bpp) = 256
At frequency of 30Mhz, each shift takes 33ns, so shifting in entire row takes 8448ns. 

8 bpp requires 8 bit planes, Bit plane 0 requires 1 cycle, bit plane 7 takes 128 cycles - total 256 cycles (2^bpp)

So shifting data for the row for all the bit planes takes 2,162,688 ns or 2.162ms. 

For 1:16 scan rate i need 16 row pair updates which will take 34.603 ms.
At this period i can achieve a 1/0.34603 frame rate = 28.9 frames per second. 

I do not see how i can chain panels, have a higher bit depth etc while i can only get 28.9 fps for 1 panel. 

I am sure something must be off in my understanding. Can anyone clear this up for me ?

hzeller, 2024-12-11:
For a 32x32 panel, there are 16 double-rows that are clocked in for each bitplane. Say there are 11 bitplanes (the default). If you assume the 30 Mhz you have 30 Mhz / (16 * 32) / 11 ... that is about 5.3kHz.

Note, each bit-plane is not clocked in multiple times, but just once, and then the time waited as needed for that bit-plane. This is why the refresh rate is much higher than you calculated. The refresh rate is limited by the on-time of the LEDs.

Waiting while the LEDs are on according to the bit plane is the most time. The lowest bit plane is on for `t`, the next for `(t << 1)`, next for `(t << 2)`...

There are a few tricks mentioned in the readme to get the higher refresh rate, but typically should should see a 100-300Hz with not much effort, and even > 1000Hz for a single bit plane.

nrichmon, 2024-12-11:
Got it! I see where I was wrong. 

So from a hardware perspective the capacity limitations will come from 1) how quickly we can write out to GPIO pins, 2) how quickly we can shift in bits to the panel and 3) the min LED on time - with 3 being the most significant capacity constraint.

Thanks

hzeller, 2024-12-11:
Also how much latency the panels introduce on the clock line - often they differ so much after a couple of panels, that things get glitchy, which might limit the length of the panel chain or clock frequency. Much more than 20-30Mhz typically does not work with panels, which is why there is a `--led-slowdown-gpio` flag.

The lowest LED on time you can control with `--led-pwm-lsb-nanoseconds`, and with `--led-pwm-dither-bits`, you can keep the few lowest bits on that time so you can tweak refresh with that a llittle.

marcmerlin, 2024-12-18:
please re-open if needed

closed 2024-12-18 by marcmerlin