hzeller/rpi-rgb-led-matrix #1747: Rpi pi4 connec RGB matrix 16x32

havituloh, 2024-12-30:
I want to ask about the RPI PI 4 connecting to a 16x32 RGB matrix, it appears but not fully, I still use manual jumper cables for the wiring, for the attached photo
[image]
[image]

davemaster, 2024-12-30:
Greetings,

Your command line please...

havituloh, 2024-12-31:
sudo ./demo -D0 --led-no-hardware-pulse --led-cols=16 --leader=32

havituloh, 2025-01-02:
Please help me

davemaster, 2025-01-03:
> sudo ./demo -D0 --led-no-hardware-pulse --led-cols=16 --leader=32

As You're using a RPi 4, you need to use --led-slowdown-gpio=<0..4> (even 5), and as those panels (16x32) are not those sell by sparkfun (32x32) you must find --led-multiplexing=<1..18> (or new values, as this library is in continue updating)

try:
sudo ./demo -D0 --led-no-hardware-pulse --led-cols=16 --led-rows=32 --led-slowdown-gpio=4 --led-multiplexing=8 (or try from 2->others)

best regards, don't forget, post a video or picture as well the command you're trying so we can diagnose the problem

havituloh, 2025-01-13:
Mr why no sucses for me

davemaster, 2025-01-13:
> Mr why no sucses for me,please help me https://github.com/user-attachments/assets/91b1bfc2-0738-4a49-aa82-949387025644
> 
> ![17367813332764259119022646954822](https://private-user-images.githubusercontent.com/44770390/402592626-b0bbc65d-c9d9-4c8f-bf7e-e8ffbce31e8e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3ODI5NzIsIm5iZiI6MTczNjc4MjY3MiwicGF0aCI6Ii80NDc3MDM5MC80MDI1OTI2MjYtYjBiYmM2NWQtYzlkOS00YzhmLWJmN2UtZThmZmJjZTMxZThlLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEzVDE1Mzc1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUxYjM1ZTBhNzE2ZjdjMzBiNGM4OGNhNzc2NDQ1YjcyYzM0ZDg5YmJhMGEzYzM1M2UwOTAzOWY5ZGE5MTY2NDgmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.fsWjZ1-457VawPmuWLRy7i8R2OTUDDkNAzSxzNpVZuY) ![17367813848962989304106758852769](https://private-user-images.githubusercontent.com/44770390/402592905-3c40a63a-5151-4901-86c5-f356a09387b0.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzY3ODI5NzIsIm5iZiI6MTczNjc4MjY3MiwicGF0aCI6Ii80NDc3MDM5MC80MDI1OTI5MDUtM2M0MGE2M2EtNTE1MS00OTAxLTg2YzUtZjM1NmEwOTM4N2IwLmpwZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMTMlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTEzVDE1Mzc1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTczNWRkOGI5OGRjZjlhNzUyOTg0YmYxZmFkOWEyMjJkNjc1M2RmNDg1YTg5Nzk4OWI1Yzc2ZTQ1MThkNDgxNmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.RWHmp3ANJK8TWb7LWOtrXuyjUns9jRhvuAu-cOlHzRM)

Hey, was you checked your command: ??? you're using cols=32, and rows=64, WRONG, do you speak English?

cols = columns = columnas -->> your panel has 32 columns
rows =rows = filas ---->> your panel has 16 rows

fix your command, and try again

havituloh, 2025-01-13:
Uploading VID_20250113_224816.mp4…

is this correct?

davemaster, 2025-01-13:
> Uploading VID_20250113_224816.mp4…
> 
> is this correct?

WHAT is VID_20250113_224816.mp4 ????

A SERIOUS advice, go read README.MD First... then come back

marcmerlin, 2025-01-13:
I'm sorry if we have to start doing this now, but there are way too many bugs opened that aren't bugs, and they are unfortunately hiding the real bugs and genuine RFEs that could be worked on.
I'll quote the first page of README:
```
If you'd like help, please do not file a bug, use the discussion board instead: https://rpi-rgb-led-matrix.discourse.group/ (obviously please read this whole page first). If you file a bug asking for personal help instead of using the discourse group, please state in your bug that you have read this entire page and that you're indeed filing a bug or request for improvement. Otherwise, please use the discourse group.
```
I'm going to close this since it does not appear to be a bug, and please continue the discussion on the help board if needed, thank you..

closed 2025-01-13 by marcmerlin