hzeller/rpi-rgb-led-matrix #1748: GPLv3 and linking from Apache-2.0 licensed programs

hudlow, 2025-01-01:
Are there any plans to relicense under GPLv3 so that programs licensed under Apache-2.0 (the permissive license [GNU itself recommends](https://www.gnu.org/licenses/license-list.en.html#GPLCompatibleLicenses)) can link to this library?

marcmerlin, 2025-01-01:
I did a bit of reading up and found
"LGPLv3 has serious issues when used with free software, and especially with the GNU GPL version 2. Projects under the GPLv2 license violate its terms if they use an LGPLv3 library (because LGPLv3 adds additional restrictions). What does FSF suggest on that case? It suggests upgrading GPLv2 projects to GPLv3"
and
"It is commonly accepted that the GNU General Public License version 2 is incompatible with the Apache License, version 2 because certain provisions of the Apache License would be considered “further restrictions” not permitted by the GPLv2"

@hzeller  I'd recommend you either dual license as GPLv2 and GPLv3, or upgrade the license to GPLv3

hzeller, 2025-01-01:
Sounds like upgrading to "GPL version 2 or any later version" should fill the bill. Will have a look tomorrow.

marcmerlin, 2025-01-03:
There you go: https://github.com/hzeller/rpi-rgb-led-matrix/pull/1751

marcmerlin, 2025-01-03:
(sorry if things are unclear, I didn't commit this to master, it's in a branch waiting for your review and merge)

marcmerlin, 2025-03-24:
let's mark this fixed.

closed 2025-03-24 by marcmerlin