hzeller/rpi-rgb-led-matrix #1749: Python Packaging

chrisgilldc, 2025-01-01:
Good afternoon!

In the effort to package some of my own python projects, it was frustrating me I couldn't include 'rgbmatrix' as a dependency.

I've worked up a fork with a pyproject.toml definition, build script, etc. I've tested it on my development Pi3 and it all seems to work fine.

Due to versioning limitations it can't be published to PyPi, but I've tested installing with pip straight from git and it works. Fork is [here](https://github.com/chrisgilldc/rpi-rgb-led-matrix) and hope it's helpful.

marcmerlin, 2025-01-03:
Hi @hzeller, I know nothing of the python bindings, what do you feel is the best course of action?
Something to potentially merge, reference, or other?
@chrisgilldc I can't guarantee it'll be merged, but can you send a PR from your tree to this project so that Henner has a chance to review and potentially, merge?

chrisgilldc, 2025-02-23:
@marcmerlin Just noting here I did create that PR back on the 5th, it's #1753

marcmerlin, 2025-11-27:
Another PR to look at if you find it helpful to integrate:
https://github.com/hzeller/rpi-rgb-led-matrix/pull/1257

ty-porter, 2025-11-28:
I think to allow us to assign you'd need to give higher permissions on the repo. I get notifications on this issue since I was tagged in the conversation if that helps.

I'm working my way through reviewing these.

ty-porter, 2025-11-30:
* #1163, #710, #1276 are all closable

* #1574 _seems_ closable but it's @chrisgilldc's issue so will defer to him

* #1619 should be closed now that #1829 is merged

* #1404 is a feature request that we need to implement

marcmerlin, 2025-11-30:
thanks a bunch @ty-porter  I closed the ones we mentioned, we're now down to 3 python issues left 🎉

rwstca, 2025-12-09:
Just wanted to mention in the `pyproject.toml` file there are personal references to the person who submitted the PR. Not sure if that is intended, just a code review catch of mine:

https://github.com/hzeller/rpi-rgb-led-matrix/blob/master/pyproject.toml#L8-L10

rwstca, 2025-12-09:
@chrisgilldc While we are doing cleanup of the Python bindings as part of this master bug thread. Is it possible to include this fix as well discussed here? This is one that is sure to continue popping up from people submitting bugs / issues.

https://rpi-rgb-led-matrix.discourse.group/t/can-not-find-imaging-h-when-doing-pip3-install-of-bindings-python/1139/8?u=deadlift89

Something like that should fix the `distutils` error for people using Raspberry Pi `Trixie` OS on Python `3.13`

marcmerlin, 2025-12-15:
Could you fine folks have a look at https://github.com/hzeller/rpi-rgb-led-matrix/issues/1840 and weigh in?

marcmerlin, 2025-12-15:
as well as https://github.com/hzeller/rpi-rgb-led-matrix/issues/1841

marcmerlin, 2025-12-16:
@ty-porter  noted. What do others here think?
As a reminder, I'm not actually a developer for this project, just a contributor like you ;)  and I've spent a bit too much time sorting and helping recently. 
My 2 goals are
1) no more bugs and PRs that get lost forever
2) trust you guys for all things python 
3) limit the amount of work for everyone

rwstca, 2026-01-11:
Any progress on the python binding Makefile conversion to hatch?

I think in the meantime we should update the readme because it's still missing mention of `python3-pil` for new installs so the python binding is effectively broken if you follow the installation guide (`Imaging.h missing error`). I think `python3-setuptools` is ok to remove reference to entirely

chrisgilldc, 2026-01-11:
I did not have cycles over the holidays. I can try to revisit the scikit approach soon, as I said, I think it's more robust for building C than Hatch - or at least I understood its docs better.
In the meantime updating the Readme makes sense.

rwstca, 2026-01-11:
> is it possible to made the board configuration an import-time option rather than compile-time option. I assume that's at least somewhat complex or it would have been done already.

I utilize this design pattern in my personal project. I store the LED matrix options in a config file in the project root folder and when I start my LED program it appends common flags (and program-specific flags) onto the command-line string of the python or C++ executable. Unfortunately it requires starting programs with a central bash script which I think people were critical of when I last proposed it.

But this is an excellent feature and would like to see it happen. If you have a brief design idea let me know and I might implement it and submit a PR for it.

zokuji, 2026-10-10:
Hello, i have detected a new bug when trying to install the python lib.
Seems like this is a syntax issue: https://github.com/hzeller/rpi-rgb-led-matrix/issues/1905

zokuji, 2026-10-10:
I've opened a PR to fix it: https://github.com/hzeller/rpi-rgb-led-matrix/pull/1906

ty-porter, 2026-10-10:
I think you beat me to it, my mistake. Looks like our CI check has a false negative now that the expected installation path is through pip.

zokuji, 2026-10-10:
I think fixing the semicolon is still not enough.
I tried pip install in local with the fix and I'm having another issue

ty-porter, 2026-10-10:
#1907 should have the full fix

zokuji, 2026-10-10:
Yes I think that's it, thank you for fixing this 👍

marcmerlin, 2026-10-10:
Thanks, just a quick reminder, that I do not test python changes, in this case it was on my cell phone while wearing a diving wetsuit between 2 dives in Fakarava :)  so please test things carefully before you send them my way, including asking for others to double review/test and telling me when it's ready for submit (I didn't ask this time around, it looked small and easy)

zokuji, 2026-10-11:
Maybe a pip install test could be implemented in CI so that we don't miss this in the future