hzeller/rpi-rgb-led-matrix #1750: Cailiang 64x32 P4 Indoor needs 64 rows but distorts output

iotricity, 2025-01-02:
I have a bunch of Cailiang 64x32 P4 Indoor panels. These panels work fine but have the odd setting they need to be set to 64 rows for a 32-row panel. Got all working except the output is distorted whatever I select in the settings. Tried different panel type chipsets, the chips used in these panels are the FM6125 but these are not available as driver.

I'm using a 3-channel board with 3 chains of 2 panels each, making it a 128x96 pixels display. Tested on a Raspberry Pi Zero 2W, also tested on a Raspberry Pi 4 2GB with the same results. If the panel is set to the physical number of rows and columns, only the bottom half of the LEDs are shown with the upper half of the chain, shifted 16 rows downwards. 
[image]
sudo ./led-image-viewer /home/pi/images/VO8W_Panel.gif --led-cols=64 --led-rows=64 --led-chain=2 --led-parallel=3 --led-multiplexing=0 --led-row-addr-type=0 --led-brightness=50 --led-no-hardware-pulse 

When selecting 64 rows instead of 32 rows, each chain is displayed correctly, but the lower row is shifted to the middle row.
[image]
sudo ./led-image-viewer /home/pi/images/VO8W_Panel.gif --led-cols=64 --led-rows=32 --led-chain=2 --led-parallel=3 --led-multiplexing=0 --led-row-addr-type=0 --led-brightness=50 --led-no-hardware-pulse

Original image
[image]

Changing the --led-multiplexing and/or --led-row-addr-type only results in one to 8 rows shown in garbled lines or checkered order distorting the image.

I've tested these panels with an ESP32 with this library [https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA](url) and things work fine when using the E-selector wire/pin and set the number of rows to 64. Also working correctly with a larger chain up to 6 panels. But I have to set the panel rows to 64 to work properly.

Anyone experienced this or knows whats wrong in my settings? Any help, hints or tips are appreciated.

iotricity, 2025-01-02:
When setting the chain length to 4 panels (i.e. double the real number of panels per row), dots get doubled in size, but all panels are used.
[image]

marcmerlin, 2025-01-03:
this does not look like a bug in the library but asking for help for unusual panels. Please use https://rpi-rgb-led-matrix.discourse.group/

iotricity, 2025-01-03:
> this does not look like a bug in the library but asking for help for unusual panels. Please use https://rpi-rgb-led-matrix.discourse.group/

Thanks for your swift reply. Posted the question in Discourse.

closed 2025-01-03 by iotricity