hzeller/rpi-rgb-led-matrix #1754: PREEMPT_RT kernel option

gregtap, 2025-01-21:
👋🏻 Hello Led lovers,

I did not see any conversation about tunning Kernel for better performance i.e increased refresh rate or less flicker interruption.

Is there some proven benefits around `PREEMPT_RT` kernel flag?

hzeller, 2025-01-21:
There might. I've tried it a couple of years back, but the result was inconclusive with various versions of the Raspberry Pi. It was a pain to compile the kernel back then so given the results I didn't bother to explore further.

Having said that, that was at least 6-7 years ago, so if someone wants to work on newer experiments, that would be great.

kjkent, 2025-03-27:
@marcmerlin From some cursory googling, it doesn't seem so, though it looks like ~ 2017 they experimented with offering it via the .txt config files: https://dietpi.com/forum/t/rpi-3-v157-rt-kernel-not-updated/1592/2.

For tweaks for rt-sensitive workloads, this is a 5-year-old post (https://dietpi.com/forum/t/dietpi-for-real-time-application/4373) but DietPi project lead Michaelng details scheduler changes that may benefit, as well as adjusting process niceness...

> Little article about what is possible: https://www.redhat.com/sysadmin/real-time-kernel
>
>However, lets see what we can do. Nice -20 is already good. But the CPU scheduling policy for the process is probably not optimal: https://manpages.debian.org/stretch/manpages/sched.7.en.html
>For real-time processes switch to FIFO (first in-first out) or RR (round-robin). If you run it as systemd service, dietpi-services can change it on service-level. To be true right now I don’t know the command to switch this outside systemd, but should be not difficult to find. The CPU priority can then be set as kinda replacement for nice.
>
>Does your process as well read and/or write to disk? In this case, the same as for CPU scheduler applies for I/O scheduler: https://manpages.debian.org/stretch/util-linux/ionice.1.en.html
Ah, its the ionice command. Use realtime for scheduling class and the I/O scheduling priority, although AFAIK that is raised automatically together with nice.
>

...Perhaps there are even better options now.

Lastly, there's https://github.com/remusmp/rpi-rt-kernel -- it builds a Raspbian Lite image with a recent kernel with RT patches applied. The build process is painless and uses Docker. For lower-spec devices it also offers 32 bit images -- as per these benchmarks (https://www.sevarg.net/2021/12/18/raspberry-pi-32-bit-vs-64-bit/), 32 bit can offer a modest performance boost over x64 for low-RAM devices. I'm using a RPi Zero 2 W, so worth a shot.

I'll pit the current DietPi I have flashed against this 32bit RT, using the cyclictest benchmark (mentioned on the rpi-rt-kernel repo) while running the D0 demo, and report back with my findings!

bluelasers, 2025-04-20:
BCM2835 based PIs will benefit from RT kernel patch. I have mixed opinions on affinity over standard multithreading. Pi is single CMT cluster. Memory stability affects some configurations more than others. (See #1468.) RTLinux in the context of multithreading is closer to affinity. (You need to demote the extra cores.)

Remove cores 1, 2 and 3 from the OS on a dedicated Pi is worth trying. Using [this](https://github.com/hzeller/flaschen-taschen) on said dedicated Pi is also worth trying. Note this will require another Pi to handle all the application logic. The benefit is your application logic will not interfere with this library. Allowing near real time performance from a non real time environment.

kjkent, 2025-05-05:
So I reflashed a Pi Zero 2 with Diet-Pi with RT kernel. I haven't had chance to do a full quantitative comparison yet, but, qualitatively, I didn't see a difference in terms of visual quality (stuttering when the refresh rate neared the max the Pi could handle) or the refresh rate itself.

It's entirely possible I'm not looking at the right metrics to judge improvement, in which case I'm happy to be informed! It could also be due to the limited resources on the Pi Zero 2 being the bottleneck, rather than the kernel.

marcmerlin, 2025-05-05:
Thanks @kjkent , even if it's not great news, a negative result is still a result.
But for your other comment, I would indeed not use anything less than an rPi3a, especially if you are using more than one output channel (I use all 3 on most of my projects).
I also have 2 bigger displays and calculating enough pixels for that framebuffer cache, seems to overflow a processor cache on rPi3 and causes visible hangs for 1 sec or more of my code updating the FB (while the display is still updating). That problem went away by upgrading to a Pi4, which has a bigger cache.

marcmerlin, 2025-05-05:
@kjkent  are you able to test again on a Pi3, or better, a Pi4?

marcmerlin, 2025-11-27:
I'm going to manually save this bug from the mass closing I'm about to do, but please send an update if you have one, or we'll close it as unresolved.

marcmerlin, 2025-11-29:
Assigning to @kjkent  for update request.  I'd indeed love to hear more of PREMPT is somehow helping but if we don't have anything definitive, I'll resolve this.

marcmerlin, 2025-12-15:
just a heads up that if this gets no updates and no one can work on it further, I will resolve this issue

marcmerlin, 2025-12-29:
@rwstca  thanks for that research.  I'm honestly happy to add 
-march=native -mtune=native -flto=2
as a default. I'll mention @hzeller here in case he wanted to vote this down if we missed something, but I think that's a good change.
did you want to send a PR to get the credit for the change and the testing/benchmarking you did?

For --led-pwm-bits=7 , that's also what I use. You lose some color depth but I find I get enough that it's the sweet spot for me.  
you will also find noticeable speed improvement from defaults.pwm_dither_bits 1 or 2 even if it makes things a bit dimmer

As for memcpy on a framebuffer, it is indeed not really an option, I'm doing this
https://github.com/marcmerlin/FastLED_RPIRGBPanel_GFX/blob/ec1ac0b8c436da7c087f6bc218de7403334ae6c2/FastLED_RPIRGBPanel_GFX.cpp#L25
I think the problem with these libs is that they need their own bitplanes that can be pushed to the displays, and the format is not compatible. 
On the plus side, those setpixel loops have been fast enough for me.

marcmerlin, 2025-12-31:
So, making things faster will help, but linux is still not a RT operating system without PREEMPT_RT.   If you really want RT this is the way. Making it go faster is good but it's still not RT.
For true flicker free, you want real RT, so I recommend you look into that and see it can be made to work.
As for your question on boot options, if you want to send a README patch to show all the optimizations you have tested, along with benchmarking numbers, I'm happy to merge that, but the lib is not going to change boot config options directly, it's for the user to do it.
And again, for -march=native -mtune=native -flto=2, I'm willing to accept a tested patch along with benchmarking numbers you already found.

rwstca, 2026-01-11:
@marcmerlin 

https://github.com/hzeller/rpi-rgb-led-matrix/pull/1855

marcmerlin, 2026-01-14:
@rwstca , I appreciate the enthusiasm. A few notes:
1) happy to accept a PR for bash runner script, it's a good idea
3) the lib can't force push a config.txt and options there will not be the same for everyone anyway, but I'm ok with your same shell script to grep desirable options there and output a "you should add/modify this in config.txt" to help the user.  Also happy to take that as README

More generally I agree that flicker reduction is a multi facet approach, I'm happy to get in all changes that will help, even if just a bit

there is also a max Hz refresh interval that helps with flicker, I can't find it right now, but I found this old issue
https://github.com/hzeller/rpi-rgb-led-matrix/issues/1689

marcmerlin, 2026-01-14:
there you go, found it: 
```
# The refresh rate will now be adapted to always have this amount of time
# between frames, so faster refreshes will be slowed down, but the occasional
# delayed frame will fit into the time-window as well, thus reducing visible
# brightness fluctuations.
#
# You can play with value a little and reduce until you find a good balance
# between refresh rate (which is reduce the higher this value is) and
# flicker suppression (which is better with higher values).
# Flag: --led-limit-refresh
#DEFINES+=-DFIXED_FRAME_MICROSECONDS=5000
```
in lib/Makefile and
./lib/led-matrix.cc:  limit_refresh_rate_hz(1e6 / FIXED_FRAME_MICROSECONDS)

./include/led-matrix.h:    int limit_refresh_rate_hz;   // Flag: --led-limit-refresh

So please play with --led-limit-refresh

rwstca, 2026-01-15:
> happy to accept a PR for bash runner script, it's a good idea

@marcmerlin @evilC  : https://github.com/hzeller/rpi-rgb-led-matrix/pull/1859

Also, would it be possible for me and @evilC to get the `Contributor` tags? We are recent contributors to the repo with our merged PRs

marcmerlin, 2026-02-10:
@evilC  on the contributor question, as I thankfully properly guessed, you were automatically added.

rwstca, 2026-03-08:
@marcmerlin sorry I stepped away awhile for life reasons.

The non-flickering updates that you made in your above comments falls apart when `taskset -c 3` fails. That is the most important part because it pins the LED process to a non-interupted core. This is exactly the underlying mechanism @kingdo9 uses in his RT-kernel approach above. Which is to have the LED process on a core with zero interupts.

So I recommend investigating (with the help of AI) why the `taskset` is failing. You should use `htop` in a second terminal window when running the LED process to make sure it's pinning to core 3.

All of the other flickering tricks are just fluff to make it slightly faster/more performant. The taskset is 80% of the magic. Chrt sets the realtime priority very high, but its meaningless if the core is still being interupted or the LED process is on the wrong core. (Or being multi-threaded and thrashed around random cores)

marcmerlin, 2026-03-12:
As for "does RT help", I now do see this on my console
```
42.834070] rcu: INFO: rcu_preempt self-detected stall on CPU
42.834101] rcu: $3-....: (3 GPs behind) idle=7d5c/1/0x4000000000000000 softirq=0/0 f
491 rcuc=5301 jiffies (starved)
42.834124] rcu: t = 525theta jiffies g = 3129 alpha = 2711 ncpus=4) 105.845757] rcu: INFO: rcu_preempt self-detected stall on CPU
105.845787] rcu: $3-....: (3 GPs behind) idle=7d5c/1/0x4000000000000000 softirq=0/0 f
915 rcuc=21054 jiffies (starved)
105.845810] rcu: ( t = 21003 jiffies g = 3129 alpha = 4721 ncpus=4)
168.858199] rcu: INFO: rcu_preempt self-detected stall on CPU
168.858227] rcu: $3-....: (3 GPs behind) idle=7d5c/1/0x4000000000000000 softirq=0/0 f
7396 rcuc=36807 jiffies (starved)
168.858250] rcu: t = 36756 jiffies g = 3129 alpha = 6251 ncpus=4)
231.870579] rcu: INFO: rcu_preempt self-detected stall on CPU
231.870609] rcu: $3-....: (3 GPs behind) idle=7d5c/1/0x4000000000000000 softirq=0/0 f
4817 rcuc=52560 jiffies (starved)
231.870631] rcu: ( t = 52509 jiffies g = 3129 alpha = 7973 ncpus=4)
242.658727] INFO: task systemd: 1 blocked for more than 120 seconds.
242.658803] Not tainted 6.12.75-rt16-v8+ #1
242.658814] "echo > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
242.659892] INFO: task (postboot): 513 blocked for more than 120 seconds.
242.659904] Not tainted 6.12.75-rt16-v8+ #1
242.659912] "echo > /proc/sys/kernel/hung_task_timeout_secs" disables this message.
```
This is probably ok and expected, but reporting just in case. Linux seems to keep running anyway.

marcmerlin, 2026-03-18:
@kingdo9  happy to reopen.
I tested on rpi3 which is very close to rpi0 2wl and the flickering went away.

As for the dmesg messages, on rpi3, honestly they are expected, I do no think it's a bug. If an RT process is running and preventing anything else from running, some other processes, including kernel processes, may get starved, and that's fine.
RT is a a zero sum game, for my display process to run at highest priority, it is going to stop some network kernel threads or whatnot to be delayed.  I do think there are some kernel warnings for stalled, and the kernel probably got stalled for short times.
I think it's fine and working as intended :)

But I'm happy to close when you ask me to since you did the most actionable work.