hzeller/rpi-rgb-led-matrix #1756: 64x64 Split Image Adafruit RGB Matrix Bonnet Troubleshooting

tidyhf, 2025-01-25:
Hi all!

I have an issue that doesn't seem to be unique to me, but looking at other threads I can't pinpoint my issues and would really appreciate a sanity check, as this is all very new to me.

I purchased an Adafruit RGB Matrix Bonnet for Raspberry Pi and have it on top of a Raspberry Pi 3 B+.

I have the bonnet connected via the barrel jack, Raspberry Pi and 1 x P3 64x64 192mm x 192mm 1/32 Scan LED panel (P-P3(2121)-32S-192x192-V8) (I want to hook up 3 more into a square formation after I get this working) all to a 5V 20A 100W Power Supply, so it's shared ground.

I have followed the official instructions doing the convenience setup and then running this command:
`sudo ./demo -D0 --led-rows=64 --led-cols=64 --led-chain=1 --led-gpio-mapping-adafruit-hat`

It then looks like the display is split into 4 sections, 1 and 3 are lit and the other 2 are just blank.  See below

[image]

At first when the ground wasn't shared and the RPi was being powered independently I could squeeze the data ribbon cable on the panel and get it to display colours across the entire panel when using gradient demo, this stopped after shared ground so I'm guessing that's irrelevant, but thought I'd mention it. I also tried to add the slow command to 1 to 4 but had no effect on the blank spaces. Video attached: https://i.imgur.com/VyZhT43.mp4

I have tried multiple panels, ribbon cables and power cables.

Adding another panel also mimics the same issue 

[image]

I have tried soldering both E and 8 and E and 16 on the back of the bonnet just in case, continuity test both times to confirm working. I've never soldered in my life so apologies in advance: 

[image]

Let me know if I'm missing anything but I would really love to get this working! Thank you so much in advance for any support.

talsan, 2025-01-29:
not sure this will be helpful, but my first soldering attempt, i just had solder over the 8, and the E just had a little on it. and experienced the same issue as you. on my second soldering attempt, both E and 8 were completely covered and it actually worked.

soldering is really hard and my hands shake like crazy.

marcmerlin, 2025-03-23:
As per readme, please don't use the bug system for support/help
you want to go here https://rpi-rgb-led-matrix.discourse.group/

closed 2025-03-23 by marcmerlin