hzeller/rpi-rgb-led-matrix #1758: small red dots appear inside every pixel on 64x64x adafruit led, with adafrui hat, and rpi 4

talsan, 2025-01-29:
[image]

does anyone know what is happening here? Let me know if I need to provide more info

raspberry pi, connected to an adafruit hat (soldiered the E and 8 together), connected to a adafruit 64x64x led matrix.

octopug, 2025-02-09:
I've got the same problem with a 32x32 Adafruit matrix. It was happening when I used 'convenience' mode and since still happens with 'quality' mode.

Btw I think you want to use 'adafruit-hat-pwm' because you did the solder job. Regardless though I still have the issue ☹️

octopug, 2025-02-11:
By 'regular' I assume you mean without the bonnet? I don't have a way to power the matrix without it sadly, so it's not something I can easily try

davemaster, 2025-02-11:
it's only a way to discard some problem with your module, your rpi gpio -it happens-

best regards.. and good luck

sub-zerohm, 2025-02-28:
Is this a p2 pitch panel?  I have those from adafruit and they don't do this.  I'm thinking maybe it's an older panel or something.

octopug, 2025-02-28:
I'm not sure which mine is exactly but [this](https://shop.pimoroni.com/products/rgb-led-matrix-panel?variant=35962488650) is where I bought it

closed 2025-11-27 by marcmerlin