hzeller/rpi-rgb-led-matrix #1774: ABC --led-row-addr-type=3 seems to require unreasonable values for gpio_slowdown

marcmerlin, 2025-03-24:
I now have 2 different kinds of ABC panels, which require unreasonable gpio_slowdown values, and the higher the resolution/more pixels being driven, the worse things get.
(splitting this off from https://github.com/hzeller/rpi-rgb-led-matrix/issues/1773 which was originally about allowing higher values than 5 for gpio slowdown)
Will also point https://github.com/hzeller/rpi-rgb-led-matrix/issues/823 to this bug

marcmerlin, 2025-03-24:
Copying things I found out in https://github.com/hzeller/rpi-rgb-led-matrix/issues/1773 , I've confirmed that on the same chip and same version of linux (I tried raspbian 32bit which was better for ABC than dietpi 64bit), when selecting ABCDE addressing (default), the output is stable with a certain slowdown factor on ABC panels, but the lines are in the wrong place.
The moment I switch to --led-row-addr-type=3, the lines go in the right place but the output becomes unstable/garbled.  So clearly the type 3 addressing line output waveform is worse than type 0 and causes the garbling.

marcmerlin, 2025-03-24:
And I took the rpi3a back to raspbian 4.x kernel (32bit) isolcpus=3, running the exact same 
Date:   Fri Feb 28 11:46:57 2025 -0800 d10f518f291bd34ca4816ef25cadf2446b882662 git 
and I can confirm that
--led-row-addr-type=0 is stable at --led-slowdown-gpio=1
while
--led-row-addr-type=3 needs --led-slowdown-gpio=4

marcmerlin, 2025-03-24:
[image]

[image]

[image]

marcmerlin, 2025-03-24:
I checked in a workaround until a proper fix can be found and will be closing the other bug I had initial data in.
I think it's good to reference to https://github.com/hzeller/rpi-rgb-led-matrix/issues/1773#issuecomment-2746713037
showing how rpi3a with an old raspbian works better than the same board with a new dietpi (exact same code)

marcmerlin, 2025-03-24:
Not sure if that adds anything, but the vendor who sold me the panels uses this to drive them: 5958 decoding

[image]

board707, 2025-03-24:
> So SM16208 must be the column chip, and gemini AI says SM5368PF are constant current chip drivers.

I think the opposite.
The SM5368 is a 595-type (ABC-shiftreg) row switcher and a SM16208 is a "common chip" constant current led driver. 

This is confirmed by the number of chips on the panel.
For SM16208 it is 8 drivers per line (128/16), multiplied by 3 colors and by two, because one set is for the upper half, the other for the lower - a total of 48 pcs.

For SM5368 it is the height of 64 divided by 8 chip outputs - 8 chips

marcmerlin, 2025-03-25:
Thanks @board707 , I didn't realize "5958 decoding" was the same/compatible with SM5368.
Ok, so gemini is wrong (which it can very much be while sounding confident), and SM5368 is the new ABC addressing row chip that seems to be subtly different from ICN2037

marcmerlin, 2025-03-25:
I checked my other A(B)C panel and it uses DP32020A row chips and that one also works but needs a gpio slowdown of 4 which seems way to high and a workaround for timing that isn't quite right.
More discussions on that chip here:
https://github.com/mrcodetastic/ESP32-HUB75-MatrixPanel-DMA/issues/645

At the end of the day, they're all shift registers, but maybe there is a small difference between them which would explain the timing issues we're seeing.

board707, 2025-03-25:
> SM5368 is the new ABC addressing row chip that seems to be subtly different from ICN2037

I don't understand what you mean, because Icn2037 is not addressing row chip at all, it is a led driver.

marcmerlin, 2025-04-03:
@board707  I've had limited time to spend on this outside of finding out why my display was going to work very badly for an event I just came back from and another one in 2 days. I'm not actually knowledgeable about all those chips and did some quick googling but may have made mistakes.
I understand what shift registers do and how they work, but the fact that there seems to be at least 3 or 4 different kinds that are not all exactly drop in compatible, is likely part of the issues we're seeing in this bug and the other that was duped against it.
(and in my case, I'm seeing this on 2 different kinds of panels with different chips with different vendors)

board707, 2025-04-03:
@marcmerlin 
In my library for STM32 there is code which according to my customers works correctly with DP32020 and SM5368 multiplexers. Unfortunately, although I have a large collection of panels, there is not a single one with such chips among them.

I could try to write a code for such a multiplexer for the HZeller library, but I need someone who will test it on a real panel with such a driver.
Another option - I can try to find such a panel, but it can take a lot of time. The problem is that many sellers do not understand at all what chips are on the panel and it is almost impossible to find a panel with a certain driver.

marcmerlin, 2025-04-03:
@board707  I can absolutely test code on the different panels I have
As to how to get ABC panels, you are correct that vendors do not understand what they are selling. I even have 10x of those ABC panels in my hands from another vendor I totally told I needed ABCDE, and then they sent me ABC anyway.
Go to this vendor https://www.aliexpress.us/item/2255801134833289.html?spm=a2g0o.order_detail.order_detail_item.3.15f1f19cT9sjc5&gatewayAdapt=glo2usa
and select the P1.875 panel ($27).  This one is ABC, I have a few of them.

thank you for offering to look into this, it is greatly appreciated.

marcmerlin, 2025-04-03:
The other ABC panels that I got 10x of without wanting them, are from a vendor I found via Email
P2 soft module 256*128mm Sunny-LEDCARD  from sales035@led-card.com
I'm only giving this for completeness if you'd like to order one for yourself ($33 plus shipping) but shipping may be high for a single panel.
I'm more than happy to test the code on those panels for you if you have new code, and if it works, I will personally review your pull request and get it merged into the main branch if Henner is not able to do so.

board707, 2025-04-06:
Hi @marcmerlin 
I added a new row switcher type to the library code. In the attached archive you will find two files that need to be replaced in the library.
After that, try to run your ABCShiftRegister panels ( with multiplexers like  DP32020 or SM5368 type), using the option `--led-row-addr-type=5`.

[abc_shift_register_switcher.zip](https://github.com/user-attachments/files/19620741/abc_shift_register_switcher.zip)

marcmerlin, 2025-04-07:
@board707 I'm actually leaving on a trip today and should be packing, but I was very excited to see your update and had to try it.
The great news is that it works!
On my rpi0 2w with 128x128 on 2 channels and passive board, I can get from slowdown 4 to slowdown 2 by changing from --led-row-addr-type 3 to 5.
Command line:
```
root@rpi02w1:~/ArduinoOnPc-FastLED-GFX-LEDMatrix# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-panel-type=FM6126 --led-rows=64 --led-cols=128 --led-row-addr-type=5 --led-chain=1 --led-parallel=2 --led-show-refresh --led-scan-mode=0 --led-pwm-bits=7 --led-pwm-lsb-nanoseconds=50 -D0  --led-slowdown-gpio=2
Size: 128x128. Hardware gpio mapping: regular
Press <CTRL-C> to exit and reset LEDs
 249.6Hz (lowest: 64.7Hz))
```
Refresh rate goes from 170 to 250! 
slowdown 1 starts showing garbage, but broken horizontally, not vertically, so the speed is now limited by the column shift registers

marcmerlin, 2025-04-07:
0Now, I went to my other board with different panels (also 128x64 ABC). This one is rpi3a with active-3 board (so, better signal coming out).
Before:
```
root@rPi3a5:~/ArduinoOnPc-FastLED-GFX-LEDMatrix/rpi-rgb-led-matrix# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-panel-type=FM6126 --led-rows=64 --led-cols=128 --led-row-addr-type=3 --led-chain=1 --led-parallel=3 --led-show-refresh --led-scan-mode=0 --led-pwm-bits=6 --led-pwm-lsb-nanoseconds=100 --led-pwm-dither-bits=1 --led-slowdown-gpio=4 -D0 
```
slowdown 4/scan 3, gave 167Hz

```root@rPi3a5:~/ArduinoOnPc-FastLED-GFX-LEDMatrix/rpi-rgb-led-matrix# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-panel-type=FM6126 --led-rows=64 --led-cols=128 --led-row-addr-type=5 --led-chain=1 --led-parallel=3 --led-show-refresh --led-scan-mode=0 --led-pwm-bits=6 --led-pwm-lsb-nanoseconds=100 --led-pwm-dither-bits=1 --led-slowdown-gpio=2 -D0 
Size: 128x192. Hardware gpio mapping: regular
Press <CTRL-C> to exit and reset LEDs
 247.6Hz (lowest: 52.0Hz))
```
slowdown 2/scan 5, now works and gives almost 250Hz

Strangely slowdown 1 gives clear line selection problems (but those panels are a different shift register). Slowdown 0 gives garbage in both directions, so slowdown 1 is the minimum required.
Please see https://github.com/hzeller/rpi-rgb-led-matrix/issues/1774#issuecomment-2746787245 for a picture of the chips

marcmerlin, 2025-04-07:
https://github.com/user-attachments/assets/a47d3dd7-10f4-455f-8018-394a2a404b08

https://github.com/user-attachments/assets/6b34e0df-19e6-4cb0-b51d-7e6dbf85c8aa

board707, 2025-04-07:
I looked, an interesting effect.
Unfortunately, I still have a poor idea of ​​how your library works and where these slowdowns are inserted so that the code works.
Next week, I hope to have a panel with a SM5368 multiplexer, then I will be able to test it in a hardware.

board707, 2025-04-07:
> the videos I uploaded above did not work in github.

Everything is showing up fine for me

board707, 2025-04-07:
> If you'd like me to send you a token of my appreciation,

Nothing to say :)
It wasn't difficult for me, I just took the ready code from my project

marcmerlin, 2025-04-07:
@board707  last test: rpi0 2w dietpi 64bit, now with active-3 board instead of passive-3:
scan 5/slowdown 2 does show some lines in the wrong place
scan 5/slowdown 1 shows the columns are very wrong but most of the lines aren't working, so there is still a sync/timing issue with that different chip as you had guessed @board707 .
Naive me thought shift registers kind of all worked the same as long as you fed the right clock.

So what we also learned is timing depends on the actual linux running and the actual version of the Pi CPU, but that active-3 boards are much more important with some, compared to others.

marcmerlin, 2025-04-07:
@board707 for planning:
1) you're going to look why SM5368 got better but is still slower than expected. Did you need help to get such a panel, or you ordered one and you're waiting for it?
2) if this is going to happen in the next few weeks, probably best if we put everything in one CL and check in your new code as one batch. If you think it will take longer, I'm happy to help check this in now
3) are you familiar with git/github and how to send a pull request to the code, or would you like me to integrate the code for you and check in as me while crediting you for the work?
4) out of curiosity, will the new shift register require yet another mapping function and be #6, or do you think you can tweak #5 to be faster on all newer panels regardless of which shift register they have?

Thank you

board707, 2025-04-07:
1. My colleague and friend has a small company that makes LED screens, I got most of my panels from him. I hope he will find me a panel with such a driver.
2. I don't think it will take more than a couple of weeks.
3. I've been using GitHub for a long time. It doesn't matter whether I create a PR or you.
4. If I understood you correctly, the answer is no - the using a new shift register doesn't need any changes in other parts of the library. A pixel mappings doesn't depends on multiplexer type.

By the way, one of my repos has a [сlassification table for HUB75 driver chips](https://github.com/board707/DMD_STM32/wiki/Led_drivers)- take a look, maybe it will be useful.

marcmerlin, 2025-04-08:
thanks @board707 I updated my question 4 above, github modified the text in ways I didn't expect.
basically I'm a bit confused as to why the shift register code doesn't just work with all shift registers and we need a scan#3, now a scan#5 and I assume an upcoming scan#6.

As for github, thanks for confirming. In that case, feel free to send pull requests whenever you are ready, and I'll review/merge them

board707, 2025-04-15:
@marcmerlin 
I received the panel  - Evo-Star 64x32 1/8 scan with SM16208 driver and SM5368 multiplex chip.

I tested it on rPi Zero with options string:

```
--led-rows=32 --led-cols=64 --led-no-hardware-pulse --led-multiplexing=1 --led-slowdown-gpio=1 --led-row-addr-type=3 
```

With `--led-row-addr-type=5` my panel works cleanly at slowdown 1, with addr-type 3 there is a strong line selecting artifacts on slowdown 1, on slowdown 2 the artifacts is almost not visible, on slowdown 3 it is completely clean. 
If you are interested in any other tests, please write.

PS:  Not related to this issue, just an observation: I also noticed that any matrix on any settings with this library blinks about once per second. There is no such effect with the same panels on other libraries.

marcmerlin, 2025-04-15:
Thanks @board707 , mine definitely have issues at all slowdowns below 4 or so, so yours and mine are not the same ;(

For the blinking, if you are using a pi0 (and not pi0 2w or other real multi core arm), the issue is not the library, but the fact that it's running linux that does a bunch of other things, some can use up the CPU for too long, and cause what you describe.
It's not as bad with multi core and isolnux=3 boot option or whatever the actual name is (it's in the README)
But ultimately a microcontroller without an actual OS will definitely work better.
Ironically the ESP32 does not quite qualify as it does run a small OS and other tasks too (although it's not as bad as linux)

marcmerlin, 2025-04-16:
@board707  fm6126, I'll admit that I send this to all my panels now since it doesn't hurt the ones that don't need it, and is required by those that do.
Panel details with chip pictures, please see https://github.com/hzeller/rpi-rgb-led-matrix/issues/1774#issuecomment-2746787245

board707, 2025-04-16:
> I'll admit that I send this to all my panels now since it doesn't hurt the ones

One moment I realized that such settings also could be a source of artifacts

marcmerlin, 2025-04-16:
@board707  are you saying that you have seen some panels behave worse after you send that init string, or you were worried that they could, but you haven't actually seen it happen?

board707, 2025-04-16:
I think it is quite likely that incorrect initialization or initialization that should not be there at all - can affect the pulse sequence on the panel.

marcmerlin, 2025-04-19:
@board707  I was able to do the test you requested, on old raspbian 32, I was able to run without FM6126 init, and it didn't make a difference, I had to use slowdown 2.
I don't have the pi0 2w with me, so I can't do the test with that (which needed slowdown 4 before)
So, I'm still running with this.  Your code is a big improvement, so I'm definitely thankful for it, but there still seems to be a sync issue with those SRs for some reason (using active-3 board on pi3a)
```
root@rPi3a5:~# ~/rpi-rgb-led-matrix/examples-api-use/demo  --led-rows=64 --led-cols=128 --led-chain=1 --led-parallel=3 --led-show-refresh --led-scan-mode=0 --led-pwm-bits=7 --led-pwm-lsb-nanoseconds=100 --led-pwm-dither-bits=1 --led-slowdown-gpio=2 -D0  --led-row-addr-type=5
```

board707, 2025-04-29:
> were you able to check your code between a 32bit raspbian on Pi3 and 64it dietpi to see the slowdown differences I experienced?

I'm sorry, I'm unlikely to be able to perform such testing in the near future. And in general, I doubt it helped in something, since the code I proposed is based on knowledge about LED panels, and not on deep understanding the RPi code; so I would hardly be able to suggest anything about the differences in the operation of different Raspberry models.

marcmerlin, 2025-04-29:
@board707  totally fair.
For the code you did right, can you send a pull request to this tree and I'll approve it, or do you want me to take your code and submit it for you?

board707, 2025-04-29:
@marcmerlin 
I would like to test another idea related to shift register type multiplexers. I will write to you in a few days.

marcmerlin, 2025-04-29:
@board707  sounds like a good plan. Please make it mapping 6, so I can benchmark 3 vs 5 vs 6 on both raspbian 4/32bit and dietpi 6/64bit, which I can do for all 3.
I'll tell you exactly what slowdown is needed for all 6 cases and we can figure out what's going on.

marcmerlin, 2025-05-07:
@board707  as an update, I can do more testing for you this week, but after that it will be difficult again for about 3 weeks.
Unless you're about to send me new code to test, I wanted to merge in your current code since I do require it for my display and I can't sync or update it until it's in git :)
Please let me know if I have your approval to merge your code into the tree under the current GPLv2/v3 license.

board707, 2025-05-07:
@marcmerlin 
I tested my idea and it didn't work. I tried adding a separate slowdown parameter only for A B C pins, in order to increase the delay only for them, and decrease the slowdown for all the others and thereby increase the panel refresh rate. 
Unfortunately, the situation did not change in my case - regardless of the duration of additional delay, my SM5368 matrix works without distortion with the main slowdown = 1 and produce a lot of artifacts with slowdown 0.
So there won't be any new code from me anytime soon.

> Please let me know if I have your approval to merge your code into the tree

yes, sure

marcmerlin, 2025-05-07:
Thanks @board707  . Now, in the "this is super confusing and I'm not sure what happened" category, I'm testing with rpi0 2w again and active-3 board and I'm able to use --led-slowdown-gpio=2 with both that chip as well as the older rpi3a.
Maybe I had bad cabling during my previous tests?
Either way, I'm now running
```
root@rpi02w5:~# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-rows=64 --led-cols=128 --led-chain=1 --led-parallel=3 --led-show-refresh --led-scan-mode=0 --led-pwm-bits=7 --led-pwm-lsb-nanoseconds=50 -D0  --led-row-addr-type=5 --led-slowdown-gpio=2
```
and getting around 250Hz refresh rate, this is pretty good.  slowdown 1 would get me 270Hz if it worked, and 0 would get me 290Hz, which is not so much faster, so the slowdown is actually pretty bearable.

with --led-pwm-dither-bits=1, I can go up from 250Hz to 350Hz, so that's a big plus and I don't really see a big difference.   

--led-pwm-lsb-nanoseconds=50 also gives a big boost of speed over the default of 100 and in the past it made my display not as bright, but right now I'm not seeing a huge difference in brightness (maybe that's more true on ABC panels than ABCDE)

Either way, the good news is that I can get up to 350Hz for 7 PWM bits (the minimum I require for my graphics) and 128x64 panels on 3 channels at the same time. Thanks again for the ABC fix, it's very much appreciated.

marcmerlin, 2025-05-07:
For my own notes and update to README later, I took the same rpi0 2w
I wanted to do a speed comparison for ABC vs ABCD.
I can confirm ABCDE is faster
```
root@rpi02w3:~# ~/rpi-rgb-led-matrix/examples-api-use/demo --led-rows=64 --led-cols=128 --led-chain=1 --led-parallel=3 --led-show-refresh --led-scan-mode=0 --led-pwm-bits=7 --led-pwm-lsb-nanoseconds=50 --led-pwm-dither-bits=1 -D0  --led-row-addr-type=0 --led-slowdown-gpio=1
```
gave me 410Hz refresh with LSB 50, and 260 with LSB 100.  Somehow 50 is not that much more dim, so these options seem to be winner:
- pwm 7 bits is enough for me
- LSB 50 gives a clear boost in speed without much visible downside in output (250Hz to 410Hz)
-  dither-bits 1 also gives a big boost from 270Hz to 420Hz without much visible output difference (2 makes things dimmer and less stable)

So ABCDE is still faster than ABC, but thanks to your --led-row-addr-type=5 , I can reach 350Hz which not 410Hz but honestly fast, and close enough

marcmerlin, 2025-05-07:
Thank you so much @board707  for your kind and valuable help, it sure made a great difference to my project and ABC panels I could barely use and can now use properly thanks to this new faster scan mode.

closed 2025-05-07 by marcmerlin

board707, 2025-05-07:
Hope for further cooperation :)

marcmerlin, 2025-05-08:
Sure thing @board707 , thanks again.
Just to confirm, for slowdown 0, on the same rPi hardware (rpi0 2w) and ABCDE panels, both vertical and horizontal output are completely broken, so slowdown 0 is just not going to happen for that hardware, whether it's ABC or ABCDE.

And just to confirm, for you the --led-row-addr-type=5 works with slowdown=1 one for you, correct? For me, 2 is required in all configurations I have for these panels (on a Pi3a and Pi0 2w, which may be faster than your Pi, I forget).
In my case, slowdown 1 looks like it would work on those panels as vertical data looks ok, but horizontal line selection is indeed broken

marcmerlin, 2025-05-08:
Slowdown 1 rpi3a active 3 board
https://github.com/user-attachments/assets/c9deab4a-a997-4141-b461-636c618b8b66

marcmerlin, 2025-05-13:
@board707  please have a look at #1788,  I figured out why you need a smaller slowdown than me.  
Can you check the bug, try my demo command line and see what refresh rate you get with slowdown 1, 2 and 3?