hzeller/rpi-rgb-led-matrix #1779: Led P5 use IC DP3265 and ICN2012, HUB 75

vanwkhanhs, 2025-04-04:
"I tried to control the P5 LED panel, but none of the LEDs light up. Has anyone successfully controlled this type of LED panel? If so, could you please share a test code that works for this panel?"

marcmerlin, 2025-04-04:
If you do not give us the command lines you tried to show you read the README in full and you tried all the relevant options, including the fm6126/7 init, you're not going to get a lot of help :)

vanwkhanhs, 2025-04-04:
int main(void) {

	/* USER CODE BEGIN 1 */

	/* USER CODE END 1 */

	/* MCU Configuration--------------------------------------------------------*/

	/* Reset of all peripherals, Initializes the Flash interface and the Systick. */
	HAL_Init();

	/* USER CODE BEGIN Init */

	/* USER CODE END Init */

	/* Configure the system clock */
	SystemClock_Config();

	/* USER CODE BEGIN SysInit */

	/* USER CODE END SysInit */

	/* Initialize all configured peripherals */
	MX_GPIO_Init();
	/* USER CODE BEGIN 2 */
	__HAL_RCC_GPIOD_CLK_ENABLE();  // Bật clock cho GPIO D
	__HAL_RCC_GPIOB_CLK_ENABLE();  // Bật clock cho GPIO B

	GPIO_InitTypeDef GPIO_InitStruct = { 0 };

	// Cấu hình PD0 - PD6, PD8 làm đầu ra
	GPIO_InitStruct.Pin = GPIO_PIN_0 | GPIO_PIN_1 | GPIO_PIN_2 | GPIO_PIN_3 |
	GPIO_PIN_4 | GPIO_PIN_5 | GPIO_PIN_6 | GPIO_PIN_8;
	GPIO_InitStruct.Mode = GPIO_MODE_OUTPUT_PP;
	GPIO_InitStruct.Pull = GPIO_NOPULL;
	GPIO_InitStruct.Speed = GPIO_SPEED_FREQ_HIGH;
	HAL_GPIO_Init(GPIOD, &GPIO_InitStruct);

	// Cấu hình PB7 làm đầu ra (LAT)
	GPIO_InitStruct.Pin = GPIO_PIN_7;
	HAL_GPIO_Init(GPIOB, &GPIO_InitStruct);
	HAL_GPIO_WritePin(GPIOB, GPIO_PIN_11, 0);

	/* USER CODE END 2 */

	/* Infinite loop */
	/* USER CODE BEGIN WHILE */
	while (1) {
		// Đưa tất cả các bit dữ liệu về mức 0
		for (int i = 0; i < 1000; i++) {
			HAL_GPIO_WritePin(GPIOD,
					GPIO_PIN_1 | GPIO_PIN_2 | GPIO_PIN_3 | GPIO_PIN_4
							| GPIO_PIN_5 | GPIO_PIN_6, GPIO_PIN_RESET);
			HAL_GPIO_WritePin(GPIOD, GPIO_PIN_0, GPIO_PIN_SET);  // clk = 1
			HAL_GPIO_WritePin(GPIOD, GPIO_PIN_0, GPIO_PIN_RESET);  // clk = 0
		}

		// Chốt dữ liệu, tắt hết LED (LAT trên PB7)
		HAL_GPIO_WritePin(GPIOB, GPIO_PIN_7, GPIO_PIN_SET);
		HAL_GPIO_WritePin(GPIOB, GPIO_PIN_7, GPIO_PIN_RESET);

		// Đưa tất cả các bit dữ liệu lên mức 1
		HAL_GPIO_WritePin(GPIOD,
				GPIO_PIN_1 | GPIO_PIN_2 | GPIO_PIN_3 | GPIO_PIN_4 | GPIO_PIN_5
						| GPIO_PIN_6, GPIO_PIN_SET);

		// Cho phép hiển thị LED
		HAL_GPIO_WritePin(GPIOD, GPIO_PIN_8, GPIO_PIN_SET); // OE = 0 (PD8) (Kích hoạt LED)

		while (1) {
			// Đẩy dữ liệu từng bước
			HAL_GPIO_WritePin(GPIOD, GPIO_PIN_0, GPIO_PIN_SET);  // clk = 1
			HAL_GPIO_WritePin(GPIOD, GPIO_PIN_0, GPIO_PIN_RESET);  // clk = 0

			// Chốt dữ liệu (LAT trên PB7)
			HAL_GPIO_WritePin(GPIOB, GPIO_PIN_7, GPIO_PIN_SET);
			HAL_GPIO_WritePin(GPIOB, GPIO_PIN_7, GPIO_PIN_RESET);

			delay(100);  // Trễ 1 giây để quan sát sự dịch chuyển
		}

		/* USER CODE END WHILE */

		/* USER CODE BEGIN 3 */
	}
	/* USER CODE END 3 */
}

vanwkhanhs, 2025-04-04:
I tested it on P10 and P5 LED panels with a 1920 Hz refresh rate, and they lit up. However, when I tried it on a 3840 Hz LED panel, it didn’t seem to work. After checking the LED controller's datasheet, I found that the data transmission method appears to be different from lower-frequency LED panels.

vanwkhanhs, 2025-04-04:
> If you do not give us the command lines you tried to show you read the README in full and you tried all the relevant options, including the fm6126/7 init, you're not going to get a lot of help :)

I tested it on P10 and P5 LED panels with a 1920 Hz refresh rate, and they lit up. However, when I tried it on a 3840 Hz LED panel, it didn’t seem to work. After checking the LED controller's datasheet, I found that the data transmission method appears to be different from lower-frequency LED panels.

marcmerlin, 2025-04-04:
this code is not using the library and you didn't provide the demo test code with command line options, which is required to help, but again you don't even seem to be using this library, so I'm closing this.

closed 2025-04-04 by marcmerlin