hzeller/rpi-rgb-led-matrix #1782: Can't find right settings for 128x128

sub-zerohm, 2025-04-16:
I have a 128x128 panel that's p1.875 that appears to be 2:128x64 panels chained together internally, however, I cannot get anything but static no matter what I pass in for panel type.  I have not tried multiplexing options yet but it may need to be mapped out.  The back has two power connectors and two hub 75 connectors, and the pinout says that it is ABC addressing.  I picked it up from AliExpress, As I've been trying random panels from to find the p 1.875s that actually have five pin addressing but these do not.  I tried all the row address types and a couple different multiplexing options as well as the panel types but I haven't had any success yet.  I will take a picture later and post it

fivemeyestore, 2025-05-10:
Your approach to this problem is very elegant. I'm learning a lot from reviewing your code.

myeyestore, 2025-05-10:
Great implementation of this feature. I particularly like the approach to error handling.

fivemhostingserver, 2025-05-10:
Impressive project structure! The architecture really demonstrates solid design principles.

eyestorefivem, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

eyesgodcode, 2025-05-10:
This is a well-structured codebase. I appreciate the clean separation of concerns.

eyesaider, 2025-05-10:
Your approach to this problem is very elegant. I'm learning a lot from reviewing your code.

fivemserverhostings, 2025-05-10:
The performance optimizations here are quite clever. Nice engineering work!

godofcodeeyes, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

eyestoreoye, 2025-05-10:
Excellent contribution to open source! The community will benefit greatly from this.

eyestorefivemstore, 2025-05-10:
Your approach to this problem is very elegant. I'm learning a lot from reviewing your code.

fivemservereyestore, 2025-05-10:
Interesting solution to this problem! I've been working on similar challenges recently.

ayestoro, 2025-05-10:
Great implementation of this feature. I particularly like the approach to error handling.

marcmerlin, 2025-05-10:
@sub-zerohm  please use the discourse group as explained in README.
I will be deleting this issue due to the spam that was added to it.

closed 2025-05-10 by marcmerlin