hzeller/rpi-rgb-led-matrix #1788: --led-row-addr-type=5 for ABC panels works with slowdown 2 with Raspbian 4 or  but with Dietpi, requires unchained panels or slowdown 3 => we need fractional slowdown factor?

marcmerlin, 2025-05-13:
First, I'll extract this important bit. addr-type=5, for 128x64 panels with the "go fast" options I'm giving below, works up to 236Hz. If I got faster, just a bit faster, because of a better compiler or kernel (with the same code), refresh rate jumps to just a bit faster, and the shift register goes out of sync and stops working.
making slowdown higher works, but fractional slowdowns would be better.

More details:
I'll fork this off from https://github.com/hzeller/rpi-rgb-led-matrix/issues/1774
@board707, type 5 is definitely better and works with a smaller slowdown, but here is what I found out

My setup is active-3 board (with level shifters), 3 outputs of 128x64:
~/rpi-rgb-led-matrix/examples-api-use/demo --led-slowdown-gpio=2 --led-panel-type=FM6126 --led-rows=64 --led-cols=128 --led-row-addr-type=5 --led-chain=1 --led-parallel=3 --led-show-refresh --led-scan-mode=0 --led-pwm-bits=7 --led-pwm-lsb-nanoseconds=50 -D0 

Here is what I found:
1) slowdown 2 works rPi3a running Raspbian 4 or Rpi0 2w, running Dietpi 6 but not rPi3a running dietpi (at all, whether panels are daisy chained, or not).

2) now I actually have 3 more panels chained one deep on each channel
P1 - panel1 - panel2
P2 - panel3 - panel4
P3 - panel5 - panel6

But this is where it gets "interesting". I do not tell the lib to output more pixels. I keep the chain at 1 and I rely on panel1/3/5 bits being shifted out by the next plane, to display on the 2nd set of panels for "free" (no more compute needed, same refresh rate).

2a) with rPi3a and raspbian 4 (kernel 4 and 32bit), the 2nd set of panels work great with slowdown 2. I get 236.0Hz, a bit slower than 2b below.

2b) same config, if I boot Dietpi 6 (kernel 6 and 64bit), both sets of panels are super damaged (the line selection is clearly wrong), whether I plug the 2nd set of panels, or not.  So the same rpi3a hardware with your same code and same everything, work with slowdown 2 on raspbian 4/kernel 4, but not dietpi 6/kernel 6.  Clearly the compiler/and or kernel do different things timing-wise and it breaks your code (maybe it's just too fast).  I get 269.4Hz, a bit faster than 2a and I guess too fast.
This is another to say that dietpi 6 is faster on that hardware and throws the signal out of spec, by a lot.
[image]

3) now I switch to rpi02wl, it gets even more interesting
3a) if the daisy chained panels are not plugged in, it works with dietpi6 and slowdown2 I get 248Hz, the display is almost ok but apparently the extra 12Hz over 2a, is just too fast if panels are daisy chained, but if they are not, the first set of panels is flawless. So another way to say this is 248Hz works, _but_ the extra noise/capacitance fed by the output ports by the mere act of daisy chaining, is causing 248Hz to be just too fast and look like this. Interestingly the mirrored panels are on the right right and look better than the original panels
[image]

3b) I switch to slowdown 3 but now the speed is down to 228Hz or so and display works.

So I think I figured out my own problem and own solution:  
--led-row-addr-type=5  helps going faster, but still hits speed limits that don't happen with ABCDE.  If we cannot go faster without upsetting the shift register logic, we need a fractional slowdown so that we can go just as fast as the hardware will allow.
I had a quick look at the code, haven't yet found where slowdown is used to do actual slowdown, only found
```
./lib/gpio.cc:  slowdown_ = slowdown;
```
not sure if slowdown can be made fractional, or not.

board707, 2025-05-13:
> not sure if slowdown can be made fractional, or not.

As far I see in current code the slowdown factor can be integer only, since it meaning is an additional loop cycles to write the same state to the output.

I would add that I also had suspicions that the reason for the interference is that the multiplexer is late in switching. But in my recent tests I tried to add separate slowdown cycles specifically for multiplexer pins and this did not prevents the interference on the panel.

marcmerlin, 2025-05-13:
@board707, 3 ideas:
1)  is it not possible to do microsleeps in the code after sending a 1 or 0?
To squeeze a bit of extra sleep, some shift registers may allow shortening how long the 0's are, or how long the 1's are (since we know the curve output by the rPi is not an ideal 0 or 1 on an analog level  (so we would have a slowdown0 and slowdown1 optoin)
1a) slowdown0
1b) slowdown1
2) Another idea is SR take bits one by one along with a clock, does it make a difference if we write the clock bit before or after changing the output bit? 
3) one more idea, still. Once all the bits have been sent, we're supposed to send a latch. Is it possible to send the bits with the clock quicker than the last signal which is to activate the latch? (basically make the latch last longer than a clock cycle)

And note that I'm not chasing something that shouldn't exist, I have the same 128x64 panels with direct ABCDE and others with ABC because a manufacturer decided to change them :( and the SR ones are twice as slow as the ABCDE ones :(

Last request when you get back to your rPi: Can you try this command and let me know how many Hz output you get?
~/rpi-rgb-led-matrix/examples-api-use/demo --led-slowdown-gpio=2 --led-panel-type=FM6126 --led-rows=64 --led-cols=128 --led-row-addr-type=5 --led-chain=1 --led-parallel=3 --led-show-refresh --led-scan-mode=0 --led-pwm-bits=7 --led-pwm-lsb-nanoseconds=50 -D0
(with slowdown 1, 2, and 3)

I'd like to see how much slower your rPi is compared to mine, or if your SR is faster than mine.

marcmerlin, 2025-05-13:
This is a bit tangential, but useful for debugging ABC also.
For 3 panels, one per chain, using the new electrodragon passive board, works with slowdown=3 on rpi02w. Full command line:
root@rpi02w3:~# ~/rpi-rgb-led-matrix/examples-api-use/demo  --led-panel-type=FM6126 --led-rows=64 --led-cols=128  --led-chain=1 --led-parallel=3 --led-show-refresh --led-scan-mode=0 --led-pwm-bits=7 --led-pwm-lsb-nanoseconds=50 --led-pwm-dither-bits=1 -D0 --led-row-addr-type=5 --led-slowdown-gpio=4
Size: 128x192. Hardware gpio mapping: regular
Press <CTRL-C> to exit and reset LEDs
 310Hz  << and I get about the highest speed I can get for ABC

It is actually stable at 310Hz and slowdown 3 (I get higher speed because I added --led-pwm-dither-bits=1 which makes the output a tiny bit less bright).
But if I add my daisy chained panels, changing nothing else, display is greatly damaged with slowdown 3 (both vertical and horizontal) simply by the capacitance, or noise that added by connecting panels on the output. Nothing else is changed.
And this happens because the input board does not have level shifters, so the signal level is 3.3V instead of 5V and I suppose is more likely to get damaged when a 2nd set of panels is added on the output, just getting the bits shifted out of free (no command line change).

The "fix" is that I need slowdown 5 with a passive board and panels on the output, which slows down normal output (dither 0) to 190Hz, but if I add --led-pwm-dither-bits=1, at the expense of a bit of brightness, I get back to 240Hz which works with slowdown 5.
Let's note that dither 0 and slowdown 5 is also an output of 230Hz refresh, but that one is pushing more data at higher speed, not directly visible on the resulting Hz refresh output, and causing the screens to display garbage if a 2nd set of panels is chained.

More of all this:
1) active over passive definitely helps pushing data faster
2) I was very surprised to confirm multiple times that simply connecting panels on the output can greatly degrade signal
3) there are tricks to get refresh speed up ( like --led-pwm-dither-bits=1 ) while sending slower output to the chips so that they don't get overloaded
4) ABC is still slower than ABCDE no matter what

MrRoy7, 2025-05-19:
Hello. I'm a newbie at the forums and have recently posted a new issue which is not excatly following the issue here in hand.
I'm just asking for any help I can get...
HUB75 RGB LED (ABC)P3.91 64x64 panel not working #1789
Thanks