hzeller/rpi-rgb-led-matrix #1789: HUB75 RGB LED (ABC)P3.91 64x64 panel not working

MrRoy7, 2025-05-14:
I'm trying to get a HUB75 RGB LED ABC panel working using the hzeller/rpi-rgb-led-matrix library on a Raspberry Pi Zero 2W using an Electrodragon HAT.. 
The panel uses MBI5124 constant-current sink drivers. I have other panels (ABCD) of the same resolution (64×64, 1:32 scan, 2-chain parallel) from a different manufacturer that work fine with this setup, **but these particular ones (ABC) won't light up at all. ** 

•	The panel works when connected to a NovaStar controller, so I know it's not defective.
•	I'm using the "regular" hardware mapping and a pixel_mapper_config of "Rotate:180;264-Mapper", along with 64 rows, 64 cols, chain_length=1, parallel=2, and multiplexing=1.
•	I've confirmed 5V and GND on the panel, and verified GND continuity between the Pi and panel.
•	I wrote a diagnostic tool that pulses all HUB75 signals individually (CLK, LAT, OE, A–E, R1–B2) directly via GPIO. No output appears on the panel.
•	I've verified cable orientation is correct (red stripe aligned with LAT/pin 1) but have not tried flipping the cable yet.
•	I implemented the recommended MBI5124 config sequence (0x7D6B), raising LE for exactly 4 clocks, and tested all reasonable combinations of OE polarity, CLK edge, LE timing, and OE blanking.

At this point I'm out of ideas. Is it possible this panel uses a slightly different MBI5124 variant or requires an undocumented init sequence? Has anyone run into a similar case where the panel works with NovaStar but not with the Pi and hzeller library?

Thanks for any suggestions.

board707, 2025-05-19:
Hi
Knowing just the type of MBI5124 chip on the panel is not enough to know if it will work.
Each panel has at least three chips. Please look up the others and let me know their id-numbers

marcmerlin, 2025-05-20:
@MrRoy7 there is a near infinite amount of panels and obviously they have to work with something or no one could sell them.  This library only supports a subset of them as no one is paid to keep adding support for incompatible panels, and those panel vendors aren't interested in helping out OSS projects like this one since it represents less than 0.1% of their sales.
If you want to use this lib, you should do the best to buy panels that are supported.

MrRoy7, 2025-05-20:
**marcmerlin**  Thanks for comments, and I agree with you. 

However I bought the ABC modules 3 years ago and had done little with them until now. I also bought two other ABCD modules at the same time, which did work. 
At the time of purchasing both sets of modules I did not understand  that the ABC or ABCD(E) style modules were in fact different, as I had been using NovaStar. I was using them for a small outdoor display.

I am now going to buy the correct modules. As I am wanting them for outdoor use I have not found the size and type I want (P3.91 64x64) that are for bright outdoor use on the usual US sites, which seem to have mainly indoor panels.
I would appreciate any contacts you have for outdoor modules using OSS lib.

board707, 2025-05-20:
@MrRoy7 
Could you please show a clear photo of the panels rear side?

MrRoy7, 2025-05-20:
**board707**  Yes I can, but unfortunately it has a molded rear cover. However I am going to carefully take the back off one, to really know what weve got !

[image]

MrRoy7, 2025-05-20:
**board707**    Here are the photos from the rear. 
As you will see I was sort of careful, but I knew I would probably have to sacrifice one board..
Thanks again for your interest and help.

[image]

[image]

[image]

board707, 2025-05-20:
1/8 scan of 64x64  - it won't be easy. It's not even a "four scan matrix", it's an  eight -scan matrix.

MrRoy7, 2025-05-21:
**board707**.    Yes I knew the panel was 8S scan (1/8) which is fairly common and what I have with the ABCD panels which run very good & bright outdoors.
As I said I expected a little damage from opening a sealed molded panel, and did not worry as I have 5 others and need only 4.

Much of the text with the chips are the manufacturers coding for assembly

board707, 2025-05-21:
> Much of the text with the chips are the manufacturers coding

Forgive my bad English, I mean a markings (id numbers, such as mbi5024 ) of chips. Please read them all

MrRoy7, 2025-05-21:
Thanks...   I'll check on the chip IDs

MrRoy7, 2025-05-21:
board707    Okay I have read all the chip details
CHIPONE  ICND2153AP  x96 :      
plus  TC7559E  x16   :   
plus /IHC245KA, NDT413000  x1

Hope this helps..

board707, 2025-05-21:
Thank you

The chips you found explain why the panel does not work with the library code.
The main driver ICND2153 on the panel is a new type, the so-called PWM with built-in memory. Drivers of this type are not supported by the library and cannot work with the current code.
As for other chips, TC7559 is a row switcher of the shift register type. To work with it you need to select a proper option
The last chips type (HC245) is a buffer

closed 2025-11-27 by marcmerlin