hzeller/rpi-rgb-led-matrix #1791: Question: Compatibility of P2.5(2020)128x64-2L64S-V2.0 panels (FM6373, TC7559KB)

markusmuc, 2025-05-27:
Hi Henner (and other contributors),

First off, thank you for the amazing rpi-rgb-led-matrix library!

I've recently acquired some LED matrix panels with the marking P2.5(2020)128x64-2L64S-V2.0.
I will be uploading/linking some photos of the panel (backside, HUB75 connector area, and identified ICs) to this issue shortly for better context.

I've identified the following key ICs on these panels:

- **Column/LED Driver: FM6373** (Datasheet: [Superchip FM6373](https://www.superchip.cn/front/product_detail?id=33&&pid=4&&name=FM6373&&typelink=FM6373%2816%E8%B7%AF1%7E64%E6%89%ABEPWM%E6%81%92%E6%B5%81%E8%BE%93%E5%87%BALED%E9%A9%B1%E5%8A%A8%29) - seems to be a 16-channel constant current PWM driver, supporting 1-64 scan)
- **Row Driver/Ghosting Control: TC7559KB** (Datasheet: [Superchip TC7559KB/QKB](https://www.superchip.cn/front/product_detail?id=35&&pid=4&&name=TC7559KB/QKB&&typelink=TC7559KB%E3%80%81TC7559QKB%EF%BC%888%E8%B7%AFLED%E5%A4%A7%E5%B1%8F%E5%B9%95%E6%89%AB%E6%8F%8F%E6%B6%88%E9%9A%90%E6%8E%A7%E5%88%B6%E7%94%B5%E8%B7%AF%20%EF%BC%89) - seems to be an 8-channel row scan/ghosting control circuit)
- **Buffer (at input): 74HC245KAJ** (Datasheet: [Superchip 74HC245K/KA](https://www.superchip.cn/front/product_detail?id=37&&pid=4&&name=74HC245K/KA&&typelink=74HC245KA%EF%BC%88%E5%85%AB%E5%90%8C%E7%9B%B8%E4%B8%89%E6%80%81%E6%80%BB%E7%BA%BF%E6%94%B6%E5%8F%91%E5%99%A8%EF%BC%89) - Standard octal bus transceiver)

My setup consists of:

- Raspberry Pi Zero 2 WH
- Electrodragon RGB Matrix Panel Drive Board (V2)

From my understanding, these chips seem to be fairly standard components for HUB75(E) panels. The FM6373 appears to be a common type of shift register/LED driver, and the TC7559KB a row driver. The "64S" in the panel name strongly suggests a 1/32 scan rate for these 64-row panels. The "2L" part is less clear to me.

Before I dive deep into testing and potentially troubleshooting, I wanted to ask:

1. Are panels with the FM6373 and TC7559KB chip combination known to be compatible with this library?
2. Are there any specific command-line flags or known quirks I should be aware of when trying to drive these P2.5(2020)128x64-2L64S-V2.0 panels? (e.g., for --led-row-addr-type, --led-multiplexing, or any specific timing parameters).
3. Has anyone in the community reported success or issues with panels using this specific chipset or this panel model?

I plan to start testing with standard parameters for 128x64 panels (likely 1/32 scan) using the electrodragon-v2-fixed GPIO mapping.

Any guidance or confirmation would be greatly appreciated!

Thanks for your time and work on this project.

Best regards,
Markus

[image]
[image]
[image]
[image]
[image]
[image]

marcmerlin, 2025-05-30:
thanks  board707

closed 2025-05-30 by marcmerlin

markusmuc, 2025-06-02:
Thank you.

I have now ordered 12 panels with FM6124 chips directly from a manufacturer on Alibaba. 

I successfully tested a panel with FM6124 chips on my "Interstate 75W" a few weeks ago without any special configuration parameters... I hope the also work with this library :)

marcmerlin, 2025-06-02:
please remember to use the discourse group instead of the bug system for general discussions, reports, progress, etc :)