hzeller/rpi-rgb-led-matrix #1794: 16x32 led matrix led mapping issue i think matrix is 4 scan

parvsonu, 2025-06-14:
Dear hzeller

I am interfacing 16x32 led matrix 4 scan and able to run but the led mapping is different so the text pattern image pattern are weird. Kindly provide your feedback. I will obliged . I have attached sample code and video of pattern i am get

https://github.com/user-attachments/assets/839edee4-c0a1-49d5-9021-b4c4d1d5a62d

ting
Number of chains:2
Number of parallel:2

My sample code:
from rgbmatrix import RGBMatrix, RGBMatrixOptions, graphics
import time
options = RGBMatrixOptions()
options.rows = 16
options.cols=32
options.chain_length = 2
options.parallel = 1
options.hardware_mapping = 'regular'  
options.multiplexing=4
options.gpio_slowdown=4
options.disable_hardware_pulsing=True
options.brightness=50
matrix = RGBMatrix(options = options)
print(matrix.width,matrix.height)
while True:
	for x in range(0, 64):
		for y in range(0, 16):
			matrix.SetPixel(x, y, 255, 0, 255) 
			time.sleep(0.100)
	time.sleep(2)

marcmerlin, 2025-06-24:
Please refer to the README and --led-multiplexing=<0..17> : Mux type: 0=direct; 1=Stripe; 2=Checkered...
If this panel does not work with any of them, then you need to look at the library code and make a new mapper

Also refer to https://github.com/hzeller/rpi-rgb-led-matrix#panels-supported

board707, 2025-06-24:
> I am interfacing 16x32 led matrix

 @parvsonu 
The panel on your video is not a 32x16