hzeller/rpi-rgb-led-matrix #1796: Compatibility question

evilC, 2025-06-24:
Hi, apologies, but I am completely new to this subject.
I am looking to build a 128x128px LED matrix.
It MUST be a flexible panel (It's to be mounted to a t-shirt)
ie something like this:
[image]
I found this item on AliExpress: https://www.aliexpress.com/item/1005003671026006.html
It's a 128x64 panel (I would use two).
The specs say that the chipset is ICN2037, which according to the link in this repo is compatible (Although the table says `Foxjone 64x32/s8 panel` next to it).
Although this library seems to only support up to 64x64 panels, it looks like this panel is maybe logically 2x 64x64 panels? - ie there are two inputs on it:
[image]
Can anyone advise if they think this would be compatible with this project?
If not, can they recommend something else?

marcmerlin, 2025-06-24:
This is a 128x64 panel either with ABC or ABCDE addressing. I have both, they both work fine, and I already have an LED outfit that uses exactly those panels.
[image]

Spoiler alert, flexible means the solder points fail and they stop working before too long (different failure modes depending on which solder points, fail).
[image]
or you can get a vertical green line, or something else

Other spoiler alert, if you are planning on using the screws behind the magnets, some are 2.5mm, which sucks, some are 3mm, which is better, but in both cases, they are glued wrong and come out
[image]
and then if you re-superglue them and put a bit too much glue, it will leak out on the pixel side on the other side and glue your pixels to whatever is touching them and then they'll get ripped out.

that said, if you don't have something to protect them, they will fall off on their own.

Closing since this is not a bug and should have been asked on discourse as requested by the README

board707, 2025-06-24:
> The height is limited to 64 pixels as there is a limit in how many to address via HUB75.

The panels with ABC type addressing ( `--led-row-addr-type=5` option ) able to control more than 64 rows.