hzeller/rpi-rgb-led-matrix #1802: Nothing displays on the panel

c0wsaysmoo, 2025-07-11:
curl https://raw.githubusercontent.com/adafruit/Raspberry-Pi-Installer-Scripts/main/rgb-matrix.sh >rgb-matrix.sh
sudo bash rgb-matrix.sh

I follow the instructions (I've done this before) I install but when I go to run a demo code like I have previous nothing displays
cd /home/xxx/rpi-rgb-led-matrix/examples-api-use/
sudo ./demo -D 1 runtext.ppm --led-rows=32 --led-cols=64 --led-limit-refresh=60 --led-slowdown-gpio=2

it says its "running" and using system resources but nothing is displayed at all. When I run it on a pi I have setup previously it works just fine. Only with a new install is it no longer working. Just happened within the last few days

closed 2025-07-12 by c0wsaysmoo

marcmerlin, 2025-07-12:
this is not a bug with the library that I can tell, but a setup problem, for which there could be a *lot* of reasons, from bad cable, bad adapter board, and so many more. You also gave no details for anyone to help you.
If you want further help, please post on discourse (see README) and give a *lot* more details on what works, what doesn't and what you tried already

c0wsaysmoo, 2025-07-13:
sorry, tried to close the issue. turns out I needed to add --led-gpio-mapping==adafruit-hat to the end of the string. just weird because I've never had to do that before. So I am not sure what changed. Either way I got it working