hzeller/rpi-rgb-led-matrix #1809: error make video-viewer - AVCodec*

jrosen73, 2025-09-04:
After updating and installing all the dependencies i receive the following error:

make video-viewer

```
video-viewer.cc:293:40: error: invalid conversion from ‘const AVCodec*’ to ‘AVCodec*’ [-fpermissive]
  293 |         av_codec = avcodec_find_decoder(codec_parameters->codec_id);
      |                    ~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~
      |                                        |
      |                                        const AVCodec*
make: *** [Makefile:40: video-viewer.o] Error 1
```

Rpi 4, ffmpeg 5.1.7, 64 bit R-pi os (Bookworm).

marcmerlin, 2025-09-04:
@hzeller  can tell you more if he has time to look at it, but my guess is that the newer ffmpeg code changed things in an incompatible way that requires this code to be changed, but then if we do so, it probably won't work with all the older/existing versions.
The path of least effort would be to use an older version of ffmpeg if you can

hzeller, 2025-09-04:
Did you update rpi-rgb-led-matrix ? That [was fixed in 2023](https://github.com/hzeller/rpi-rgb-led-matrix/commit/043a52921d237341af1c24028eca40b17400d716).

closed 2025-09-04 by hzeller