hzeller/rpi-rgb-led-matrix #1812: C# Bindings - File Not Found Microsoft.Win32.Primitives

scottparise, 2025-09-30:
When compiling and trying to run the C# bindings examples, i get this error:

Unhandled exception. System.IO.FileNotFoundException: Could not load file or assembly 'Microsoft.Win32.Primitives, Version=9.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'. The system cannot find the file specified.

File name: 'Microsoft.Win32.Primitives, Version=9.0.0.0, Culture=neutral, PublicKeyToken=b03f5f7f11d50a3a'
   at System.ConsolePal.EnsureInitializedCore()
   at System.Console.add_CancelKeyPress(ConsoleCancelEventHandler value)
   at Program.<Main>$(String[] args)

I've updated the csproj to use net9.0 and both the RPiRgbLEDMatrix.csproj and the example project build without issues, only when you try to run i get this error. 

 dotnet --list-sdks
9.0.305 [/root/.dotnet/sdk]

dotnet --list-runtimes
Microsoft.AspNetCore.App 9.0.9 [/root/.dotnet/shared/Microsoft.AspNetCore.App]
Microsoft.NETCore.App 9.0.9 [/root/.dotnet/shared/Microsoft.NETCore.App]

everything seems to be installed correctly

Can anyone help with this issue?

scottparise, 2025-09-30:
I installed .net 6 just to see if it worked. I was able to build each project, but when i try to run, i get this error:

dotnet run
/root/.dotnet/sdk/9.0.305/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.EolTargetFrameworks.targets(32,5): warning NETSDK1138: The target framework 'net6.0' is out of support and will not receive security updates in the future. Please refer to https://aka.ms/dotnet-core-support for more information about the support policy.
/root/.dotnet/sdk/9.0.305/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.EolTargetFrameworks.targets(32,5): warning NETSDK1138: The target framework 'net6.0' is out of support and will not receive security updates in the future. Please refer to https://aka.ms/dotnet-core-support for more information about the support policy.
/root/.dotnet/sdk/9.0.305/Sdks/Microsoft.NET.Sdk/targets/Microsoft.NET.EolTargetFrameworks.targets(32,5): warning NETSDK1138: The target framework 'net6.0' is out of support and will not receive security updates in the future. Please refer to https://aka.ms/dotnet-core-support for more information about the support policy.
Suggestion: to slightly improve display update, add
        isolcpus=3
at the end of /boot/cmdline.txt and reboot (see README.md)
Unhandled exception. System.DllNotFoundException: Unable to load shared library 'libSystem.Native' or one of its dependencies. In order to help diagnose loading problems, consider setting the LD_DEBUG environment variable: liblibSystem.Native: cannot open shared object file: No such file or directory
   at System.Random.XoshiroImpl..ctor()
   at System.Random..ctor()
   at Program.<Main>$(String[] args)

marcmerlin, 2025-10-01:
I'm not sure anyone is maintaining those. If anyone who cares and knows enough about C# can submit patches to fix/improve this, they will be very much welcome